LongSoft/UEFITool #13: Can't open Lenovo G505s bios

hexum, 2015-01-08:
Hi!  
  
I've just compiled UEFITool and tried to open Lenovo G505s bios file with no luck. An error message "parseBios: One of volumes inside overlaps the end of data" occurred.    
What's the reason? Could you help me?  
[Link to BIOS file](https://github.com/hexum/G505s_no_whitelist/blob/master/original/WinVALGC300.bin)

NikolajSchlej, 2015-01-09:
The reason is simple: this binary file is not a bios, but a PE32 application with a BIOS region image inside. UEFITool is meant to work with BIOS images and I have no plans to add any support of some proprietary BIOS update formats like that one, because:
1. their numbers are many
2. some vendor updates it's format every once in a while and it needs updating the tool
3. most of formats can be unpacked using external tools (your case)
4. modified images of vendor formats aren't working as good as originals because of checksums, digital signatures and other things like them

Your particular case is named "InsydeFlash unpacked" and it's a PE32 app with BIOS region inside. The region can be located by searching the "$_IFLASH_BIOSIMG" ASCII signature, followed by two UINT32s: FullSize and UsedSize, which normally has the same value of 0x400000.
Long time ago I wrote a small utility to extract BIOS images from this type of binaries, you can download it here: http://forums.mydigitallife.info/threads/48979-UEFITool-UEFI-firmware-image-viewer-and-editor?p=884343&viewfull=1#post884343

NikolajSchlej, 2015-01-11:
I'm closing this issue because I have no plans to integrate such functionality directly into UEFITool for the reasons mentioned before.

closed 2015-01-11 by NikolajSchlej

hexum, 2015-01-11:
Anyway thanks for good tips.
But I still see misleading error message...
I think adding file format conformance checking before dissection could be a better practice.

NikolajSchlej, 2015-01-11:
This error message is true: at the very end of this PE32 file (offset 0x5726B8) there is a block with strings, which contains UEFI volume header signature "_FVH". Right before the signature there must be UINT64 FvLength, the length of the volume ("11 D6 7E 2F 17 C0 9A 10" in the file), which is obviously too long for the the rest of the file (0x200). That is why UEFITool shows the "Invalid UEFI volume " error and a more specific "parseBios: One of volumes inside overlaps the end of data" message. It's not misleading in any way as for me, and it's exactly a part of format conformance checking, which that BIOS file failed to pass.

hexum, 2015-01-11:
I have to agree with you, this behavior is acceptable.
After the error raising UEFITool showed me partial tree and I started to think that problem in the file ending. I didn't realize the file in not a UEFI bios at all.