LongSoft/UEFITool #18: Enhancement request: support DB of well-known GUIDs

jethrogb, 2015-02-09:
It would be nice if you could load a file with GUIDs and descriptions (or maybe this would be compiled in) and the UI would display the description along with the GUID.

Here's a script that will scrape GUIDs from .dec files, from e.g. [TianoCore EDK2](https://github.com/tianocore/edk2):
https://github.com/jethrogb/uefireverse/blob/master/dec2c.rb

NikolajSchlej, 2015-02-10:
It's on my TODO list, but I don't see it as a major feature that needs to be done immediatelly. File GUIDs are changing and can't be collected without reverse-engeneering lots of vendor-specific images, and most files has UI section with their names anyway. PPI and Protocol GUIDs are rerery needed in UI (shown only in DEPEX-sections), and can be just googled if you need to know what a specific GUID represents. Anyway, it's on the list and someday I will do it for sure.

XVilka, 2015-04-14:
Can be easily done by the user itself if embed python scripting support with exporting some simple api with GUIDs an so on exposition.

NikolajSchlej, 2015-04-14:
I will add a plug-in system and/or scripting support, but not until all planned major changes (i.e. new builder code) are done and tested. I've already decomposed the FFS engine into separate parser/finder/operations/builder classes so every one of it can be extended separately (possibly by scripting API in a not-so-near future) or thrown away if not used at all.

XVilka, 2015-04-14:
By the way, not sure about python scripting, since it is a 'de facto'  standard in RE world, but embedding [duktape](http://duktape.org) would be much easier.

NikolajSchlej, 2015-04-15:
It a bit of a problem for me because I know neither Python nor JS good enough to support them on any good level. Anyway, this is a future project and now I have to finish that builder already before implementing any fancy, because the whole tool is just a viewer without it.

skochinsky, 2016-07-04:
I second this request. At times there are images without UI names and if the DB is a simple txt file, it shouldn't be too difficult to let users post additions as pull requests or even GitHub issues. Here's a small collection of filename/section GUIDs I've collected which can be a start: https://gist.github.com/skochinsky/3b8f948bfef6f8b32c0577ac018bdaf9

You could also add a feature where if file(s) with unknown GUID but present UI name are encountered, the tool automatically prompts the user to submit them

NikolajSchlej, 2016-07-04:
I agree, time to return to this feature. I'm now working on getting console-based tools independent from Qt library, so they can be used in non-GPL environments, wrapped into a lib and so on. It's about 50% done and I don't have enough time to dedicate myself to it now, but I also expect this feature to be done soon, together with hex viewer.

Cr4sh, 2017-03-09:
This feature was implemented in latest versions of new_engine branch, you can specify your own GUID's database using "File" -> "Load GUID database...".
Binaries are here: https://github.com/LongSoft/UEFITool/releases/tag/A40

skochinsky, 2017-03-21:
Nice! I would suggest either shipping the default DB in a separate file or adding an export option, since right now the only way to figure out the file format is to look into git.

Velocet, 2017-04-27:
I put togehter two GUID database files. Both work with A40.
- The first file contains the official GUIDs taken from the .dec and .inf files found in EDK2 2015
  - [EDK2_2015_GUIDs-2017-04-27.csv](https://gist.github.com/Velocet/d394281d96191e235ff46a8aa2018d80#file-edk2_2015_guids-2017-04-27-csv)
- The second file is a mixture of the "old" GUID database, @skochinsky's [gist file](https://gist.github.com/skochinsky/3b8f948bfef6f8b32c0577ac018bdaf9) and the file mentioned above
  - [UEFI_AIO_GUIDs.csv](https://gist.github.com/Velocet/d394281d96191e235ff46a8aa2018d80#file-uefi_aio_guids-csv)

I adapted the [bash script from Laszlo Ersek](http://feishare.com/efimail/messages/20140307-0247-_edk2__script_to_translate_GUIDs_to_names-Laszlo_Ersek.html) and used some Excel (:trollface:) to get rid of most duplicates, etc.:
```bash
#!/bin/bash
set -e -u -C

unset LANG || true
export LC_ALL=POSIX

{
  # Grep the DEC files for GUID definitions, and remove everything
  # except the GUID name stem and the GUID bytes.
  git grep -h -E '^ *g[A-Za-z0-9_]+Guid *= *\{' -- '*.dec' \
  | dos2unix \
  | sed -r \
        -e 's/^ *g([A-Za-z0-9_]+)Guid /\1 /' \
        -e 's/[={},]/ /g' \
        -e 's/ +$//' \
  | while read -a INPUT; do
      NAME=${INPUT[0]}
      unset INPUT[0]
      printf '%08X-%04X-%04X-%02X%02X-%02X%02X%02X%02X%02X%02X:%s\n' \
          "${INPUT[@]}" "$NAME" 2>/dev/null
    done

  # Search all .inf files.
  git grep -E -H '^ *\<(BASE_NAME|FILE_GUID)\> *=' -- '*.inf' \
  | dos2unix \
  | sort \
  | {
      FILE_PREV=
      MACRO_PREV=
      VALUE_PREV=
      while read FILE MACRO EQ VALUE COMMENTS; do
        if [ x"$FILE_PREV" = x"$FILE" ] \
            && [ BASE_NAME = "$MACRO_PREV" ] \
            && [ FILE_GUID = "$MACRO" ]; then
          printf '%s:%s\n' "$VALUE" "$VALUE_PREV"
        fi
        FILE_PREV=$FILE
        MACRO_PREV=$MACRO
        VALUE_PREV=$VALUE
      done
    }

  # Some well-known GUIDs that used not to be available from the above
  # sources.
  printf '%s:%s\n' \
      1810AB4A-2314-4DF6-81EB-67C6EC058591 BootScriptHeaderData \
      605DAB50-E046-4300-ABB6-3DD810DD8B23 ShimLock_MokOwner \
      9A8D3433-9FE8-42B6-870B-1E31C84EBE3B BootScriptExecutorImage \
      AEA6B965-DCF5-4311-B4B8-0F12464494D2 BootScriptData \
      B5AF1D7A-B8CF-4EB3-8925-A820E16B687D BootScriptDataOrg \
      DEA652B0-D587-4C54-B5B4-C682E7A0AA3D AcpiS3IdtrProfile

  # We'll eliminate duplicates below. The above generation order means priority
  # order too.
} | sort -f -u -t : -k 2,2 --stable
```
The script itself won't work but if you copy every part it works... don't have time to hassle around with bash.

NikolajSchlej, 2018-02-10:
I think this can be closed because DB is already there since A40.

closed 2018-02-10 by NikolajSchlej