LongSoft/UEFITool #20: PEIM PE body replacement leading to duplicate file

johnrandomsmith2, 2015-02-16:
I was just testing whether I could modify a BIOS binary with UEFITool and write it back to the machine with an SPI programmer. I just randomly picked a PEIM, PchSpiPeim out of this HP BIOS. I right clicked the PE32 within it and did "Extract Body" and saved out the PE32 file. Then I opened it in CFF explorer, went to the end of the .text section, and changed some padding 0s to 0x90s. Then I went back into the file, right clicked on the PE and said "replace body" and selected the modified PE with nops at the end. But when I do this, UEFITool starts displaying 2 PE sections under that GUID. I can see it's set to rebuild the file, but when I save it out and open it again, there are still two PE sections under the same PEIM file GUID. Am I doing something incorrect? Despite the extra file, the BIOS is still bootable. It's just that I don't know if I made a meaningful change if it would actually be using the proper file.

[image]

The HP BIOS I'm modding is available here:
https://drive.google.com/file/d/0B-fqKYB_fAF-aFQ0TW1BN2pweTA/view?usp=sharing

(While I'm submitting I should also say that when I try to replace a DXE file within the compressed volume, it seems to do the replacement correctly, but the resulting BIOS is un-bootable once written back to the HP. I just don't know how I could help debug this for you... But if you have any thoughts feel free to email me at my username at gmail and we can guess and test things.)

NikolajSchlej, 2015-02-16:
[image]
There is no editing for the tree elements (because they are immutable by desing), so every edit is a combination of marking old item for removal and new ite, for insertion. In this case it's got marked for Insert and then immediatelly for Rebase, because it's a PEI file that needs to be based properly for PEI phase to work. I think you hide your Action column somehow, so you don't see the actions, and then marked the first item for any other action (i.e. Rebuild), so both of old and new PEI modules ended up there. I ried to reproduce the problem reported, but it works OK for me, even if work with exactly the same BIOS image and FFS file.

As for DXE volume modification - the problem is known and solved, please read my post [here](http://habrahabr.ru/post/249655/), use Google Translate if you need.

NikolajSchlej, 2015-02-17:
Checked it again in all three OSes, can't reproduce, so I'm closing it for now.

closed 2015-02-17 by NikolajSchlej

johnrandomsmith2, 2015-02-17:
Hmm, I think the problem must have been because I was using an old version of UEFITool. I thought I had recreated the problem with the latest (and the problem did exist, here's an example picture showing the Action column is clear on a resulting image). But I just re-tried on both Windows and Mac with v 0.20, and it worked fine on both. Sorry I didn't properly re-test with the latest before posting.

Thanks for the pointer to the DXE volume post though!

[image]