LongSoft/UEFITool #23: Add support for additioinal PE32+ image types

apop2, 2015-05-13:
The latest pecoff v8.3 specification from microsoft:
https://msdn.microsoft.com/en-us/windows/hardware/gg463119.aspx

AARCH64 (ARM64) images are already running UEFI firmware
#define IMAGE_FILE_MACHINE_AARCH64        0xAA64

In the EDKII build tools, there are some references to PPC: (file BaseTools\Source\Python\CommonDataClass.py)
EBC | IA32 | X64 | IPF | ARM | PPC | AARCH64

#define IMAGE_FILE_MACHINE_POWERPC	0x1f0	//Power PC little endian
#define IMAGE_FILE_MACHINE_POWERPCFP	0x1f1	//Power PC with floating point support

NikolajSchlej, 2015-05-13:
Useless without real images to test on, but will add anyway because it's trivial.

NikolajSchlej, 2015-05-14:
Done in new_engine branch, thank you for report.

closed 2015-05-14 by NikolajSchlej

apop2, 2015-09-21:
To aid in your testing, below is a link to an AARCH64 compiled firmware image:
http://builds.96boards.org/snapshots/hikey/linaro/uefi/latest/fip.bin

Also, there will be a PPC64 port of UEFI that will be pushed out on github soon.

NikolajSchlej, 2015-10-02:
Tested the image, thanks. Works without anz issues on new_engine branch:
Type: 10h
Full size: 22A24h (141860)
Header size: 4h (4)
Body size: 22A20h (141856)
DOS signature: 5A4Dh
PE signature: 00004550h
Machine type: ARM64
Number of sections: 4
Characteristics: 002Eh
Optional header signature: 020Bh
Subsystem: 000Bh
Address of entry point: 260h
Base of code: 260h
Image base: 0h