LongSoft/UEFITool #24: Add support to display microcode revisions

apop2, 2015-05-13:
Add the ability to parse and display the microcode versions that are contained in the UEFI firmware image.

NikolajSchlej, 2015-05-13:
Very hard to implement properly because each IBV implements it differently and I can't write it for AMI's case because of my NDA. 
Some new images have Firmware Interface Table, that points directly to each microcode binary, so it can be done with ease. Will think about it.

Cr4sh, 2016-11-02:
Microcode version feature was implemented in Alpha35:  https://github.com/LongSoft/UEFITool/commit/4160a6a58023f65daf17c05926bc148a009ae0b1
Currently it supports only microcode blobs located in Firmware Interface Table.

timofonic, 2017-11-01:
@Cr4sh So why is this issue still open? :confounded: 

What are the missing items?

skochinsky, 2017-11-01:
In addition to FIT, you can also scan BIN files/RAW sections/padding areas for microcode headers, they're pretty distinct. See e.g. https://github.com/platomav/MCExtractor and https://github.com/ddcc/microparse
AFAIK pre-FIT most vendors just stuffed all updates concatenated in a block of flash covered by a ROM hole or a pad/raw binary FFS file.

vit9696, 2018-11-12:
Fixed previously, as of NE A51.

closed 2018-11-12 by vit9696