LongSoft/UEFITool #25: Ability to see subsections of the PE32 images

apop2, 2015-05-14:
PE32 images, in uefi, can contain multiple sections.  The sections of a PE32 image follow the EFI_IMAGE_SECTION_HEADER, and are contained immediately after the optional PE32 header.

UEFI drivers compiled with the "UEFI_HII_RESOURCE_SECTION = TRUE" will create a .rsrc section that containsii the Hii String, IFR packages, etc.

Being able to extract, and replace the .rsrc subsection would allow modification of  setup string, etc.

NikolajSchlej, 2015-05-15:
It will also allow more specific search results. Will add this to the TODO list, thank you for report.

NikolajSchlej, 2016-07-09:
Tried to implement it properly, but it seems to require a whole PE rebuild code for builder to work properly, so I decided to abandon the feature. Use any suitable PE editor instead.

closed 2016-07-09 by NikolajSchlej