LongSoft/UEFITool #26: Video bios extraction

victoredwardocallaghan, 2015-06-12:
Hi,

I noticed the too shy post and wanted to express this, so this issue is more of a question. Last time I tried to use UEFITool it was not obvious how to extract the video card blob. If you have not already, could a option be implemented that makes this task much more straight forward?

Thanks,
Edward.

NikolajSchlej, 2015-06-12:
Hi,

it can be done for sure, but will require a bit of CSMCORE parsing, that I can't legally implement from scratch because of my NDA with AMI. 
The easiest solution right now is to use UEFIExtract to extract all image contents into a folder, and then any tool suitable for searching in multiple binary files (bgrep, for example) to search for a file containing VBIOS. 
I will thin about it anyway, perhaps I could implement it without any NDA breaks after all.

NikolajSchlej, 2015-06-12:
Created #27 with this proposal as initial one.

cyrozap, 2015-11-22:
There are some simple methods [here](http://www.coreboot.org/VGA_support#UEFI_Method) which I wouldn't think would violate any NDA's.

NikolajSchlej, 2015-12-09:
I don't think this special feature belongs to UEFITool or UEFIExtract, more like to a script that parses UEFIExtract output (.dump directory) to find any files the user want. 
Will try to convince someone to write it. :)

cyrozap, 2015-12-10:
Wow, I completely forgot UEFIExtract existed... In that case, I agree, we can just use tools like `grep` to find what we need. As a simple test, running `grep -r --include="body.bin" -l "VGA Compatible BIOS" "3 BIOS region" | grep "0 Raw section"` successfully found the VGA BIOS for my laptop.

NikolajSchlej, 2015-12-12:
Good, I'll close this issue, because the problem is solved and I won't dig into this further.

closed 2015-12-12 by NikolajSchlej