LongSoft/UEFITool #33: AMI and an apple

lordkag, 2015-09-18:
I see you have closed my issues. They were so young and beautiful, will be endlessly missed. Now I'm going to open others, as a revenge. Take that, human!

- I've tested the new UEFIFind and it works as expected. What I can recommend is to also add an empty line after "nothing found" and replace "uefitool" with "uefifind" in the help message. On my old laptop it takes around 30-40 seconds for 80 mixed patterns on the hardest images, but after that it is a breeze for Extractor. It is even faster for most images and way faster than before. On powerful desktops it should be in a blink of an eye.

- I take it that you only show the FIT when there is a VTF? They say an apple a day keeps the doctor away, but [this one] (https://support.apple.com/kb/DL1823?locale=en_US) will cause more pain and frustration than actually help. I believe you will need a Linux or OS X to unpack the .pkg inside, as on Windows I could only do it with AnyToISO. After that navigate to EFIPayloads and take MBP121_0167_B07_LOCKED.fd as an example. Unlike the .scap images, this one looks (almost) flash-ready. There is a FIT structure, despite UEFITool empty tab. At first I thought it is just not linked to VTF, but on closer inspection, it is in a strange way. Check in Non-UEFI_data.pad, it is made of two identical blocks of 0x20000 bytes. Each block contains 0x8000 zero padding + 0x4400 microcode + 0x4000 microcode + 0x7C00 zero padding + 0x30 FIT + 0x1FD00 zero padding + 0x6000 Volume. That volume has a VTF that links to the second FIT. Why would they do that and what are your plans? Honestly, I wouldn't want to be in your shoes when you receive such reports. On one hand, you want a nice and clean tool, with only the specs to worry about and some nice additions once in a while. On the other hand, doing so may prevent some images being fully recognized or properly assembled, therefore extra checks and checks for the checks are needed, adding more complexity to the model. 

- From the same package, take any scap images, like MP61_0116_B15_LOCKED.scap. In the capsule information, the full size and image size are equal. The Intel capsule is also affected, while AMI and Toshiba ones are showing the proper values.

- Speaking of capsules, let's talk about AMI. Do you have her number, is she into robots? If not, compare the images from [ASRock Z97 Extreme 6] (http://www.asrock.com/mb/Intel/Z97%20Extreme6/?cat=Download&os=BIOS) and [Asus SABERTOOTH Z97 MARK 1] (https://www.asus.com/Motherboards/SABERTOOTH_Z97_MARK_1/HelpDesk_Download/). I suppose any ASRock or Asus image would do. I haven't yet read about the structure and I'm not even sure there are public sources, but shouldn't AMI capsule be made as GUID + UINT32 ROM offset + UINT32 Flags + UINT32 Full size + UINT16 Header size + UINT16 Map offset + Certificate + ROM Map/Layout? The certificate appears to be the same as in Intel signed sections: UINT32 size + UINT16 revision + CertificateType (WIN_CERT_TYPE_EFI_GUID) + CertificateGUID (EFI_CERT_TYPE_RSA2048_SHA256_GUID) + CertificateData (EFI_CERT_BLOCK_RSA_2048_SHA256 = HashType (EFI_HASH_ALGORITHM_SHA256_GUID) + PublicKey[256] + Signature[256]). The map area is made of UINT64 Address + UINT32 Offset + UINT32 size + UINT32 type + UINT32 attributes, from a quick read. Because I see some contradiction on those images. The Asus one has the ROM offset pointing to 0x1C, while the ASRock to real image offset. The Asus image only covers the first hash in certificate size, while ASRock covers both hashes and padding until ROM map. There is also the fact that ASRock capsule declares the full size as 0x601000, when the image is 0x801000. Maybe they meant to target just the BIOS region, but how is that possible? The MAP area already has the hooks for signing whatever blocks you want. And how is UEFITool showing the full size as 0x801000?

NikolajSchlej, 2015-09-19:
Hi LK, took that, still standing. :wink: 
- both of your recommendations are implemented in UEFIFind 0.10.1 commit
- it appears that Apple images do have a normal VTF at the end, so now i'm just search backwards in each non-UEFI data block inside a volume and if there is a VTF at the end of it - if will be added in the tree. This will make builder code just a bit more messy, but since it's not even ready yet, I have no fear.
- corrected together with the previous one in UT NE A12 commit, thanks for spotting it. Will correct for 0.21.x branch soon.
- as for AMI capsules and stuff, you already know how I see any image elements that aren't written into SPI flash - "Padding". The sole purpose of UEFI capsule header from UEFITool's point of view is to tell where the real image starts, and it fulfills it for now. 
As for full size, I just show file size in this field, because I've already seen that there can be some crap in that field.

lordkag, 2015-09-29:
I believe you can close this issue. I will open a new one, just to keep you busy. I'm such an evil robot!

closed 2015-09-30 by NikolajSchlej