LongSoft/UEFITool #34: To padd or not to padd

lordkag, 2015-09-29:
Plutomaniac sent me an image that showed some errors in Extractor. Upon inspection, I noticed it was because of a little bug in UEFITool engine. To test it, just take any Intel SPI image and add something after BIOS region with hex editing. This is what you will get:

- That padding can't be extracted properly. It will have the right size, but it will be extracted from the end of BIOS region.
- There is no search in that padding.
- If you search something that is present in BIOS region right where UEFITool places this padding, you will get a false result, as if it was found in that padding.

I believe they are all due to the same error, the offset of the padding is wrong.

NikolajSchlej, 2015-10-01:
Thanks again for reporting, evil robot.
It was all the same bug, that is now corrected, but I can't reproduce the offset problem. 
Can you provide a file with incorrect padding offset?

lordkag, 2015-10-01:
Upon further reading, my report is not accurate enough without an example. Let me provide a "bugfix" for my report, you provide a bugfix for UEFITool, then we can split the fortune in half. Evil robot laugh preparing to launch... "Are you sure you want to launch evil laugh?"... Press "Yes". "Something happened :("... Damn you, MicroRobot!

- Take an Intel image with regions and 800000h in size. Add "I'm such an evil robot. I'm such an evil robot!!" at the end. Size of padding is 30h.
- If you open the file in UEFITool, this padding is placed after and outside BIOS region, as expected. The offset is 800000h, still in order. From a visual point of view, everything is OK. Let's move to processing.
- Try to search "evil robot". Nothing found.
- Try to extract added padding. It will be extracted from [BIOS_Region_End - Padd_Size -> BIOS_Region_End]. Offset of padding is wrong.
- Try to search some bytes from [BIOS_Region_End - Padd_Size -> BIOS_Region_End]. You will get one/more matches in BIOS Region, but also one for outer padding. Offset of padding is wrong.

Conclusion: Placement and offset is right in visual mode, wrong in processing mode. Solution: give me your money!

NikolajSchlej, 2015-10-01:
Aha, now it's all clear and all solved already. There wasn't any problems with the offset, the problem was that the padding (which is just "x bytes at the end") was gotten not from the whole image, but from the BIOS region. It's solved already for both branches, so I will close this issue once I release the new binaries.

closed 2015-10-02 by NikolajSchlej