LongSoft/UEFITool #35: I'm on your tail

lordkag, 2015-09-30:
Take this [Intel file](https://downloadcenter.intel.com/downloads/eula/14364/BIOS-Update-CL94510J-86A-?httpDown=https%3A%2F%2Fdownloadmirror.intel.com%2F14364%2Feng%2FCL0034P.BIO).

- File F5CF71AB-B04B-4B7E-988A-D8A0D498E692 is not reporting Data checksum as invalid, but it gets fixed anyway after saving the file.
- Some files have FFS_ATTRIB_TAIL_PRESENT and EFI_FFS_FILE_TAIL is present in those. UEFITool reports the wrong size (minus tail) and extracts the file with reported_size + 1 (or tail cut in half). If I try to change a byte in such a file (example 140BAF0E-8D12-478F-A0B4-17B9B7B5CEED), it fixes the tail, but tail is shifted to right by 1 byte, so it results in first byte of old tail being included in file next to new tail. This time, the size of file is accurately reported (if you ignore the fact that it added an extra byte), but it gets extracted as reported_size - 1. Volume GUID 7A9354D9-0468-444A-81CE-0BF617D890DF is revision 1, shouldn't all this be supported in UEFITool?
- File 4A538818-5AE0-4EB2-B2EB-488B23657022 is a mess, all the sizes are wrong, even after removing Intel custom data at top and bottom. File C62B4EFE-9147-4589-8DEA-ADB0173A4441 can be fixed with removing Intel extra data (obviously not recommended), but it has duplicate files. These two files will most likely be used to prepare the file for flashing, nothing you should worry about.

NikolajSchlej, 2015-10-01:
Corrected the extraction for new_engine branch (there were no tails at all), but will get to master branch again in 1-3 days.

NikolajSchlej, 2015-10-02:
Got it done a bit earlier, please test and close the issue, if it's really fixed.

lordkag, 2015-10-02:
Yes, it seems fixed from what I tested. What do you mean by "close the issue"? Do I have the power to lock and unlock issues, power over UEFILand, one robot to rule them all? Finally! Here are some of the changes I plan to implement: you will still have an active role in developing new and interesting technologies (you're fired!); users will be pleased to have a larger set of options (cash or credit); the legacy will continue the inspire the future (name will be changed to UEFIBend); enthralling new features (remote connection to webcam and other juicy spywares). First decree: this issue is closed and sentenced to everlasting oblivion

closed 2015-10-02 by lordkag