LongSoft/UEFITool #36: Report Generation Capabilities

apop2, 2015-10-01:
It would be nice to be able to generate a report about the current BIOS and save the report to a text or comma separated value file.  This would be useful when trying to compare two versions of a BIOS when trying to determine the individual FFS files, ME regions, etc, that have changed between the two versions.

XVilka, 2015-10-01:
Imho, that is a task for a plugin. So implementing something like Python scripting into the UEFITool would allow to implement things like reports generations etc.
Also porting 'subzero' tool on the top of the UEFITool would be awesome.

NikolajSchlej, 2015-10-02:
I'm solving the problem of comparing two BIOS versions by using UEFIExtract and BCompare, but it's for sure not an option for mass comparisons.
Do you have any ideas on what that report should look like?

apop2, 2015-10-02:
I figured the generated report could mimic the treeview that UEFI tool uses for displaying a BIOS, but with a few additional fields.  Additionally, position information, such as offset from the start of the information, could be ignored.

As an example:

Intel Image
  Descriptior region
  ME region
  BIOS Region
    <FV GUID>
      <FFS file GUID> <size of FFS file>
        <section> section size
        <section> section size

Using the report generated for the rom image would make it very easy to compare two versions of ROM image to see which FFS files changed, or which regions of the bios changed.

NikolajSchlej, 2015-12-09:
Will return to this soon, got some ideas on how it should look like to be helpful

NikolajSchlej, 2016-06-25:
Done as part of UEFIExtract 0.12.0, I'm waiting for your feedback

apop2, 2016-06-27:
The report format is outstanding. I ran it on two different versions of a bios and compared the two and I was able to see which components were modified between the two.  Marking as closed.

closed 2016-06-27 by apop2