LongSoft/UEFITool #38: UEFIExtract folder names in OS X

gdbinit, 2015-11-10:
Hi,

UEFIExtract will not use the GUID for firmware volumes for OS X EFI images. It instead uses "AppleCRC32 AppleFSO" text field that shows ip on UEFITool.

Instead of this code in FfsEngine::recursiveDump
QString childPath = QString("%1/%2 %3").arg(path).arg(i).arg(model->text(childIndex).isEmpty() ? model->name(childIndex) : model->text(childIndex));
it probably should be
QString childPath = QString("%1/%2 %3").arg(path).arg(i).arg(model->name(childIndex));

Personally I find it more useful to have the GUID as folder name instead of that text.

Best,
fG!

NikolajSchlej, 2015-11-10:
Thanks for reporting, I will add a workaround to show GUIDs for firmware volumes even if their text fields aren't empty. Patch is ready, new binaries will be done tomorrow.

NikolajSchlej, 2015-11-25:
Please excuse me for not building those binaries, I will be in Russia up to the 6th of December, will build them after returning home.
You can build a binary yourself, the required PRO file is in "UEFITool/UEFIExtract" of new_engine branch.

gdbinit, 2015-12-05:
Ooops got lost on my mental queue. Don't worry, I got UEFITool building in OS X (pretty easy, great work!) and created something for what I wanted to do using UEFITool core code.
Keep up the great work :-)

NikolajSchlej, 2015-12-12:
Will do. Closing this issue now, will release a never version today after restoring the build environment lost to a broken SSD.

commit 5fd8edf0 (2015-11-10): Fix #38

- added a flag to use FV GUIDs as folder names even if they have
non-empty text fields. Thanks to OsxReverser for #38.
closed 2015-12-12 by NikolajSchlej