LongSoft/UEFITool #41: Whiplash on you

lordkag, 2016-01-28:
What a better way to start the new year, than with a bug report? Get back to work, lazy human, bring me the ultimate tool for slashing UEFI!

- I've noticed a new Lenovo capsule E20BAFD3-9914-4F4F-9537-3129E090EB3C in [this file] (https://download.lenovo.com/consumer/mobiles/dbcn22ww.exe). Use Universal Extractor or InnoExtractor, then BEEF or /cap or /ext to get the BIOS.cap file.

- Why do I get UEFIFind error code 8, when doing a mass search on [this file] (http://www.mediafire.com/download/ka6t3dodap1tedr/20151014_originalBIOS.rar)? It's not limited to this file alone, just a sample. The common ground seems to be when there is nothing found in any of the patterns. The weird part is that I get this error code when capturing the output with Python, thus UEFIFind displays the result, but somehow returns error code 8.

- In [this file] (http://www.mediafire.com/download/eocpft7ewpypd8p/FC0027.rar) there is a decompression error for those "non-UEFI data found in sections area". UEFIRomExtract can decompress it.

- I'm getting a QString error when using UEFIExtract on Intel signed sections.

- Another Lenovo capsule 25B5FE76-8243-4A5C-A9BD-7EE3246198B5 in [these files] (http://www.mediafire.com/download/g37vd8p0v0ebcl0/lnvgy_fw_uefi.rar), also again UEFIFind error code 8 with CAPSULE_A.scap.

- In the same archive, for file SPSFullME.bin, is there no check for the space between regions? As you can see, that file is using one of the reserved regions to map the space between 0x1000 and 0x11000.

- Not an actual bug, just rounding to magic number 7 and checking your devotion to the robot cause. [Can you feel me?] (https://www.youtube.com/watch?v=hfzws2YBy9M&feature=youtu.be&t=50)

- Wait, I lied. When opening images like SPSFullME.bin, with no volumes or bios region, no offset is shown for rest of the regions. I don't expect the fancy fields (memory address, compression, fixed) to be present, but the offset should be known and displayed. In 0.30.0_A18 I'm not getting an offset at all, in any image. But this might be due to my building tools, as I mentioned in the PM.

NikolajSchlej, 2016-01-28:
Hi @lordkag, long time no see. :smile: 

Let's get it backwards this time: 
* the offset problem is known and it's made this made this way because alpha isn't called alpha for no reasons. Actually, it's done differently from A16, I'm trying to add this information recursively right now, and it seems I've failed to add a call to this new addOffesetsRecursive() function, it will be solved in A19 for sure, I promise.
* No, there is no check for space between regions right now, and now I finally have an image to convince myself to implement it properly, thanks a lot for finding it.
*  For other stuff - will investigate both Error 8 and QString issues, and add this new GUIDs to the list of known ones.

NikolajSchlej, 2016-01-29:
@lordkag, you mighty robot, I need your wisdom now.
- The reasond behind returning 8 aka ERR_ITEM_NOT_FOUND is in this code:
`if (found.isEmpty()) return ERR_ITEM_NOT_FOUND;` for the single item search,

 and in this code: 'if (!somethingFound) return ERR_ITEM_NOT_FOUND;' for a multi-search. This second check will return 8 only if nothing were found, and 0 otherwise.
Please tell me what is an expected behavior here and I can adjust it accordingly.

- I've tested the interaction between UE and Intel signed sections, but haven't found any QString errors on my files. Could you please provide a file and steps to reproduce the problem?

- Detection of proper decompression algorithm for EFI_STANDARD is a PITA. The strategy now is to try Tiano, and if it fails, try EFI1.1, but here we have some small sections that can be decompressed "correctly" (i.e. the size of decompressed data matches the indicated size in section header) using both algorithms and the only way distinguish between them is to parse both and select the most suitable algorithm based on parsing results. This is a complication, but I really don't see any other options here. If you have some - please share your thoughts.

- Detection of free space between regions will be done, it's also not too easy because the order of regions can be messed up, but I will try to get it working ASAP.

- All other stuff is already done, I will push the changes and build new binaries in a pair of days.

lordkag, 2016-01-29:
- The single item search should stay as it is, it is useful for batch processing (or any processing based on error code), to know what was the search result and act accordingly. For instance, if I search for a GUID, I need a precise result/error-code, to know if I can extract that GUID. On the other hand, I don't see the error-code (related to search result) to be a necessity for multi-search. What would be the reason for that? If nothing found, do what? One of the reason for a multi-search is to capture that output and process the result. If someone uses the multi-search from the command prompt, the error code is irrelevant, it has the result printed on screen; if someone uses the multi-search in a batch, the error code is needed only for general errors, not for the search result. For instance, I use Python to capture the output, as can be seen in UEFI_Strip -> ZyxWorker.py -> "mass" option. Python treats any non-zero error code as an error and throws an exception. I added a catch for the exception, but if UEFIFind returns ERR_ITEM_NOT_FOUND, I can't capture the valid search result, at least not in a pretty way. You might argue that having no positive result for any of the search patterns means there is nothing to process in the end, but I should have the option to capture that output, just like it is printed on the screen. I see two solutions for this: either you comment that "if" statement in multi-search - I tested this and it works - leaving the end user to deal with the output; or you leave it as is and I change my code to something like `if error_uf.returncode == 8 : print("Nothing was found with UEFIFind!")`. In the end, it depends on what is more useful for the end user: the error code or the result itself. If you want to keep the current code, it is perfectly fine for me, I just have to change one print from a warning to a notice. In fact, this might actually mean less work for my script, because it wouldn't need to process the output if all searches returned nothing, it will be redundant. One question though, I suppose "found" variable remains True is one positive result is found, not the result of the last search. I don't doubt your code, I doubt my ability to read your code.

- Take [this file](http://www.mediafire.com/download/f7z6www6wbiah5j/Intel_Signed.rar). You will see the errors on UEFIExtract 0.10.6, last released version. Related to this, is the warning "parseSection: signature may become invalid after any modification" for Intel signed sections not present in latest UT 0.30.0 releases, or it wasn't there at all from the beginning?

- I think you should add the parsing method of distinguishing EFI/Tiano compression only to small sections, where the decompressed size can coincide. Not that I know anything about these compression algorithms, but I doubt they will output the same size for bigger sections. The question is, what is the magic number: 0x1000, 0x5000, ...?

- Not sure if you noticed, but you should push the changes in two stages. Start by treating that region as padding and check if you can detect padding between regions, add it to the tree in the right position, extract it correctly. The second stage is to add reserved regions to the tree, when found. I did this already in my ZyxWorker -> "ireg" option (largely based on UEFITool), because I didn't trusted OEMs to respect the Reserved tag. The result [can be seen here](http://www.mediafire.com/view/7zsqu6fa6ye7ept/Intel_Regions.png)

NikolajSchlej, 2016-02-01:
- I will leave it as is then, because it's consistent and you can simplify the parsing a bit if nothing is found.

- Solved already, will commit and release a fixed version soon.

- A better solution is to try one algorithm, check if it's unpacked to the same size, if not - try another one. If both have the same size - parse both, select a meaningful data. If both aren't meaning full or both have incorrect size - it's an error and should be reported.

- This reserved regions must be parsed as reserved and added as reserved, no doubt here, but there can also be a padding between regions (FPT likes to tell about it if you try to flash such a file), and now it's also not in the tree and therefore such images can't be properly rebuilt. Will dome something with it ASAP.

NikolajSchlej, 2016-02-02:
Solved all issues in new_engine branch, will backport the changes to master in a pair of days.
New versions of UEFITool_NE, UEFIExtract and UEFIFind will be released now.
@lordkag, you have all the power to close this issue after testing them.

lordkag, 2016-02-04:
What is this? I asked for the buttons to be "Dark imperial blue", not "Dark midnight blue". As mentioned in our agreement - page 26272, paragraph 34, line 12, note 3, detailed on a 2 microns space bellow that page - all your claims are void if you don't fulfill my demands. Surrender your project and live with the eternal shame!

I will close this issue and you don't have to upload a new fix, chances are no one will notice until the next release. But remember to add it on the list for the next release: the file with Tiano/EFI conundrum is almost fixed. The content is retrieved, but it is added twice in the tree.

NikolajSchlej, 2016-02-05:
Contents added twice is actually an issue to be solved, so I will reopen it for a day or two.

NikolajSchlej, 2016-02-09:
Finally solved, new releases will be done this week, hopefully.

closed 2016-02-09 by NikolajSchlej