LongSoft/UEFITool #44: Dependency Graph

RafaelRMachado, 2016-03-29:
First of all this project is amazing!

Would it be possible to add a dependency graph option ?
I think on something to let easier to understand what modules are dependent to each other.

NikolajSchlej, 2016-03-29:
Thanks.

There is a problem with dependency graph, and it's a big one: modules are very rarely dependent on modules (the only cases are BEFORE and AFTER opcodes). Most of modules are dependent on PPIs/protocols, but there is no easy way to determine what module publish/unpublish/republish a given protocol. There are some easy heuristics like "if the module has protocol GUID inside, but it's not in it's DEPEX section, it's probably publishing it", there are some mode complex heuristics, there are disassembler/decompiler engines that can be used to locate calls of RegisterProtocol/NotifyProtocol and similar functions - possibilities are many, but all of them are very hard to automate well enough.
One of the best options is symbolic execution, like @jethrogb did [here](https://jbeekman.nl/blog/2015/03/reverse-engineering-uefi-firmware/), but it also very hard to do without human assistance. And then there are events and notifications, and callback, and interrupts, and SMI handlers, and... 

I don't say the feature in't worth it, but there are many easier things to do right now, like NVRAM parsing/validation and new builder code, so I don't think I can get into DepGraph problem any time soon.

jethrogb, 2016-03-29:
> "if the module has protocol GUID inside, but it's not in it's DEPEX section, it's probably publishing it"

In my experience, even this is not always the case in practice. For example the ThinkPad modules I've worked with often contain many GUIDs inside which are never referenced by the code.

NikolajSchlej, 2016-03-29:
Yep, seen such cases too. 
DepGraph based on firmware image only will be a tough nut with symbolic exec/virtualization, and even tougher without it, that's why the only thing done in this general direction was DEPEX parsing.

NikolajSchlej, 2022-08-24:
Because there are already way better tools doing a similar thing, like [efiXplorer](https://github.com/binarly-io/efiXplorer), we will not pursue this as part of UEFITool.

closed 2022-08-24 by NikolajSchlej