LongSoft/UEFITool #46: reserved identifier violation

elfring, 2016-04-09:
I would like to point out that identifiers like "[`_FLASH_DESCRIPTOR_HEADER`](https://github.com/LongSoft/UEFITool/blob/19cc0312445145bbde46333bfab593e45d770da2/descriptor.h#L22)" and "[`__UEFITOOL_H__`](https://github.com/LongSoft/UEFITool/blob/63e5a4dd1cf187fe6b62bd0da60529746f2ff7a1/uefitool.h#L14)" [do not fit](https://www.securecoding.cert.org/confluence/display/cplusplus/DCL51-CPP.+Do+not+declare+or+define+a+reserved+identifier "Do not use identifiers that are reserved for the compiler implementation.") to the expected naming convention of the C++ language standard.
Would you like to adjust your selection for unique names?

NikolajSchlej, 2016-04-09:
Thanks, will sure fix it, just wan't aware about this convention.

NikolajSchlej, 2016-04-09:
Fixed in new_engine, which will be the new master branch after FfsBuilder code will be ready. 
I don't want to fix it in the current master, as it's not a bug and doesn't require a new binary release.

closed 2016-04-09 by NikolajSchlej

elfring, 2016-04-09:
How do you think about to [delete misplaced underscores](https://stackoverflow.com/questions/228783/what-are-the-rules-about-using-an-underscore-in-a-c-identifier#answer-228797 "What are the rules about using an underscore in a C++ identifier?") also from [affected data structure identifiers](https://github.com/LongSoft/UEFITool/blob/80a2d57f62694b13ae919192ffba205df62b7605/common/descriptor.h#L21 "Update candidate")?

NikolajSchlej, 2016-04-09:
Done in https://github.com/LongSoft/UEFITool/commit/d6909fe9b6e92d957715d43046091540e5ea0ffb.

elfring, 2016-04-09:
Thanks for [another small source code improvement](https://github.com/LongSoft/UEFITool/commit/d6909fe9b6e92d957715d43046091540e5ea0ffb "Renaming of some data structure identifiers").