LongSoft/UEFITool #48: Remove an unnecessary null pointer check

elfring, 2016-04-09:
[An extra null pointer check is not needed](https://isocpp.org/wiki/faq/freestore-mgmt#delete-handles-null "Do I need to check for NULL before delete p?") in [a function like "UEFITool::init"](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/uefitool.cpp#L92 "Update candidate: UEFITool::init()").

NikolajSchlej, 2016-04-09:
There is no **unneeded** checks there. This check is required to prevent calling operator delete on NULL pointer if init() is call right after UEFITool object creation.

Optimizations like this are a) pointless for almost any program that doesn't crunch terabytes of data and b) dangerous, as you may not fully understand an intended behavior before calling a check "unnecessary".

closed 2016-04-09 by NikolajSchlej

elfring, 2016-04-09:
Would [a corresponding software update](http://dietmar-kuehl.de/mirror/c++-faq/freestore-mgmt.html#faq-16.8 "Do I need to check for NULL before delete p?") be more useful for [the function "UEFITool::saveImageFile"](https://github.com/LongSoft/UEFITool/blob/1a6d2142b0fa1685fb0091804661cfa4214eee59/UEFITool/uefitool.cpp#L672 "Update candidate: UEFITool::saveImageFile()")?

NikolajSchlej, 2016-04-09:
I stand corrected, doesn't know it's safe to call operator delete or free() on NULL. Checks removed in new_engine branch.

elfring, 2016-04-09:
Thanks for [your small source code improvement](https://github.com/LongSoft/UEFITool/commit/ba0779b4156306a91b8bd4f8e0c3cc7177c1869c "Deletion of a few unnecessary checks").

DeadMG, 2016-04-15:
That's not really an *improvement*. You've taken some shitty deletes and made them trivially less shitty. Smart pointer or go home.

NikolajSchlej, 2016-04-15:
Will go home than.