LongSoft/UEFITool #50: Cannot compile any other tools with make command

laptander, 2016-04-19:
I am trying to create PKGBUILD for Arch Linux, but I cannot do this, until I solve problem of compiling.
UefiTool itself is not a problem, compiling normally.
But UefiExtract fails to compile:
    
    qmake-qt5 uefiextract.pro
    make
`g++ -Wl,-O1 -Wl,-O1,--sort-common,--as-needed,-z,relro -o UEFIExtract uefiextract_main.o ffsdumper.o types.o descriptor.o ffs.o ffsparser.o peimage.o treeitem.o treemodel.o utility.o LzmaDecompress.o LzmaDec.o EfiTianoDecompress.o moc_treemodel.o   -lQt5Core -lpthread 
ffsparser.o: In function `FfsParser::parseNvarStore(QByteArray const&, QModelIndex const&)':
ffsparser.cpp:(.text+0xfd4a): undefined reference to `nvarAttributesToQString(unsigned char)'
ffsparser.o: In function `FfsParser::parseVssStoreBody(QModelIndex const&)':
ffsparser.cpp:(.text+0x1c559): undefined reference to `efiTimeToQString(EFI_TIME_ const&)'
ffsparser.o: In function `FfsParser::parseFlashMapBody(QModelIndex const&)':
ffsparser.cpp:(.text+0x1f133): undefined reference to `flashMapGuidToQString(EFI_GUID_ const&)'
collect2: error: ld returned 1 exit status
Makefile:340: recipe for target 'UEFIExtract' failed
make: *** [UEFIExtract] Error 1`

Also for UefiFind

    qmake-qt5 uefifind.pro
    make
`
g++ -Wl,-O1 -Wl,-O1,--sort-common,--as-needed,-z,relro -o UEFIFind uefifind_main.o uefifind.o types.o descriptor.o ffs.o ffsparser.o peimage.o treeitem.o treemodel.o utility.o LzmaDecompress.o LzmaDec.o EfiTianoDecompress.o moc_treemodel.o   -lQt5Core -lpthread 
ffsparser.o: In function `FfsParser::parseNvarStore(QByteArray const&, QModelIndex const&)':
ffsparser.cpp:(.text+0xfd4a): undefined reference to `nvarAttributesToQString(unsigned char)'
ffsparser.o: In function `FfsParser::parseVssStoreBody(QModelIndex const&)':
ffsparser.cpp:(.text+0x1c559): undefined reference to `efiTimeToQString(EFI_TIME_ const&)'
ffsparser.o: In function `FfsParser::parseFlashMapBody(QModelIndex const&)':
ffsparser.cpp:(.text+0x1f133): undefined reference to `flashMapGuidToQString(EFI_GUID_ const&)'
collect2: error: ld returned 1 exit status
Makefile:340: recipe for target 'UEFIFind' failed
make: *** [UEFIFind] Error 1
`
The corresponding line numberrs are: 

- [3132](https://github.com/LongSoft/UEFITool/blob/new_engine/common/ffsparser.cpp#L3139) for nvarAttributesToQString
- [4225](https://github.com/LongSoft/UEFITool/blob/new_engine/common/ffsparser.cpp#L4325) for efiTimeToQString
- [4643](https://github.com/LongSoft/UEFITool/blob/new_engine/common/ffsparser.cpp#L4746) for flashMapGuidToQString

Also, where did UefiPatch disappeared from new_engine branch?

NikolajSchlej, 2016-04-19:
Sorry for not updating the .PRO files of UEFIExtract, will do it now. Add nvram.h and nvram.cpp there and try again.
I'm now debugging some NVRAM-related quirks and will release a working UEFIExtract afterwards.
As for UEFIPatch: it requires the FFSBuilder to be ready, and because the engine code is rewritten, older version won't work anyway, so I removed it till the new builder code is ready.

laptander, 2016-04-20:
Thanks, it worked. Just created a pull request, that adds nvram to pro files.

NikolajSchlej, 2016-04-20:
Thanks for the PR, merged.

closed 2016-04-20 by NikolajSchlej