LongSoft/UEFITool #52: Expose FIT in Structure so UEFIExtract can extract it

timevortex, 2016-05-18:
Hi,

I really like how the UEFITool UI in new_engine displays the FIT table but this information seems to get lost when I run UEFIExtract on the same flash image. Can you please look at exposing this within the structure so it will get extracted out to disk with UEFIExtract?

I've attached a screenshot here. As you can see in this example, the FIT table physical address show that it's within the Padding area. When I extract this out, I just get body.bin and info.txt for the Padding area volume and the details of the FIT table aren't exported out in any manner.

Cheers,
Parth

[image]

NikolajSchlej, 2016-05-18:
Will do. The focus for UE now is to remove dependency from Qt, so please don't expect this change to be done tomorrow, but it's rather easy change and will be done eventually.

timevortex, 2016-05-18:
No worries :+1:

NikolajSchlej, 2016-06-20:
Done. Is it OK for FIT to be shown to stdout instead of a file?

NikolajSchlej, 2016-06-22:
I thinks it's fine to have it this way after all.

commit ee3a2562 (2016-06-20): UE 0.10.9

- added FIT table output to stdout, closes #52
closed 2016-06-22 by NikolajSchlej