LongSoft/UEFITool #53: Trailing newline missing in info.txt files

timevortex, 2016-05-18:
Hi,

I've noticed that the info.txt files that get dumped to disk by UEFIExtract don't have a trailing new-line for the last line of the file. This creates all sorts of interesting issues for Linux tools and python which all expect [POSIX compliance](http://pubs.opengroup.org/onlinepubs/9699919799/basedefs/V1_chap03.html#tag_03_206) for all lines in a file.

I'd like to help you remedy this by sending a patch to new_engine. I'm opening this issue to discuss what your preferred option of fixing this is. From what I see there are three ways to fix this problems and I list them below. Obviously each option has its advantages and disadvantages. Options 3 seems the easiest and most future proof as it doesn't require you to change how you code but it feels like a hack :)

Anyway, please let me know which option you prefer (or if you have another option I didn't list) and I'll send a pull request to you!

Cheers,
Parth

## Option 1 - correct existing code

Go through [ffsengine.cp](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/ffsengine.cpp) and make sure that every occurrence of `info += tr` or `info = tr` that appears just before a call to `model->addItem` has a `\n` as the final character of the text.

An example of doing this would include changing the current [line 240](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/ffsengine.cpp#L240) to have the extra `\n` added:
`    QString info = tr("Full size: %1h (%2)\n")`

This is tedious but the real downside is if you append new info later then you have to remember to make sure the `\n` stays in the correct place.

## Option 2 - add a new line of code before addItem

Go through [ffsengine.cp](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/ffsengine.cpp) and add `info += tr("\n")` just the line before every `model->addItem` call.

An example would be adding a new line at [line 243](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/ffsengine.cpp#L243) that will state:
`info += tr("\n")`

This feels like repetitive coding.

## Option 3 - patch addItem directly

Modify in only the one central place by changing the [addItem function in treemodel.cpp](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/treemodel.cpp#L271) so that the first code at [line 275](https://github.com/LongSoft/UEFITool/blob/d54f215e673d1dc338ef3171e82a220376d40c9a/treemodel.cpp#L275) reads:
`info += tr("\n")`

## One example of the problem

As you can see the shell gets wrapped wrongly:
```
pparth@doctor:~/bios.bin.dump/4 FFF12B8D-7696-4C8B-A985-2747075B4F50/2 Free space$ cat info.txt 
Type: Free space
Subtype: 
Offset: 5D1000h
Full size: 1F000h (126976)
Compressed: No
Fixed: Yes
Memory address: FFDD1000hpparth@doctor:~/bios.bin.dump/-2747075B4F50/2 Free space$ 
```

timevortex, 2016-05-18:
Sorry, I just realised that all the code links above reference the master branch - the concepts remains the same for new_engine so I won't bother updating the links.

timevortex, 2016-05-18:
I just tried option 3 for testing and unfortunately it's adding an extra `\n `in the middle of info.txt so this may involve a bit more digging by me. I'm happy to do it though once you give me an indication of your general preference.
```
pparth@doctor:~/bios.flash.dump/3 BIOS region$ cat info.txt 
Type: Region
Subtype: BIOS
Offset: 800000h
Full size: 800000h (8388608)

Compressed: No
Fixed: Yes
Memory address: FF800000hpparth@doctor:~/bios.flash.dump/3 BIOS region$
```

NikolajSchlej, 2016-05-18:
Add '\n' at the end of [this line](https://github.com/LongSoft/UEFITool/blob/new_engine/UEFIExtract/ffsdumper.cpp#L71) and it's solved.

timevortex, 2016-05-19:
:+1: Works perfectly!

Thanks for quick fix.

closed 2016-05-19 by timevortex