LongSoft/UEFITool #58: problem in lenovo t460

jawal37, 2016-07-14:
i have issu in bios lenovo t460 win i make chang in module laptop npt work 
i have teste all version uefitool
i have ather lenovo t450 i can mak modification withwot problem uefitool work gret
i think problem in compression of module 
thinks for your help 
if you want bios dump for t460 i can send it and for t450 if you want compar bitwin
sory for my english

NikolajSchlej, 2016-07-14:
You PC is protected by Intel BootGuard technology, as you can see by opening it with UEFITool NE and looking at the last 2 entries of FIT table. Because it's a HW-based technology, and I don't have Lenovo's private key, I can't make the modified BIOS work no matter what.
INTENDED/WONTFIX

jawal37, 2016-07-14:
thinks for your replay
yes i have see that
but in t450 i can make change and he have bootguard to

Closed #58<https://github.com/LongSoft/UEFITool/issues/58>.

-

NikolajSchlej, 2016-07-14:
BootGuard being active or not is determined by chipset/SoC fuses blown or spared during motherboard manufacturing. It's impossible to say how a random PC will behave without trying. If your T450 works after modifications - there are many possiblities, ether BG is not enabled, or Lenovo failed to implement chain of trust properly, or anything else. Anyway, it's not a fault of UEFITool and can't be solved.

jawal37, 2016-07-14:
no no i respecte your work you do nice work i now your tools is good i have solde with it many of problem
befor i work with your i have test phonex tools and he work just for t440 in t450 he dont work but your tools work for t450 and t440
i want just now how it work
thinks and sory

NikolajSchlej, 2016-07-14:
No problem, sorry if my reply seemed offensive to you.

jawal37, 2016-07-14:
thinks

NikolajSchlej, 2016-07-14:
Hi again, sorry for too fast (and probably incorrect) response.
I will prepare a test versions with modified LZMA compression (lower compression level compatible with current EDK2's LzmaCompress.c), please test it.

NikolajSchlej, 2016-07-14:
Try this version: http://rgho.st/6qVhjdQKc

jawal37, 2016-07-15:
thinks i teste it and i t ell you

you now problem i ts in comprison of modulle win you make change in it

i am not good in tools but with your tools i have mak chang in t450 but phonxie tools not work for this 5em geniration just 4em geniration

thinks

UEFITool_test.zip<http://rgho.st/6qVhjdQKc>
rgho.st
property=

-

jawal37, 2016-07-15:
no seem problem not work
thinks
if you want i send you dump for
t450 and t460 if you want compar betwin

NikolajSchlej, 2016-07-16:
You can send the both. Please also send your modified images that work for t450 and don't workd for t460.

jawal37, 2016-07-16:
http://rgho.st/6BT5hWK8y
t460.7z<http://rgho.st/6BT5hWK8y>
rgho.st
property=

jawal37, 2016-07-16:
http://rgho.st/6xBmZYPQ8

t450.7z<http://rgho.st/6xBmZYPQ8>
rgho.st
property=

NikolajSchlej, 2016-07-18:
Studied the images, BG seems disabled in t450 and enabled in t460. UEFITool isn't to blame, no tool can modify that image without having Lenovo private key.

closed 2016-07-18 by NikolajSchlej

jawal37, 2016-07-18:
thinks nikolaj