LongSoft/UEFITool #59: Unable to build UEFIDump after 59a6f2 commit

laptander, 2016-07-15:
You have deleted fitParser.h and fitParser.cpp in 59a6f2 commit.
But you did not edit UEFIDump/uefidump.h. [UEFIDump/uefidump.h](https://github.com/LongSoft/UEFITool/blob/new_engine/UEFIDump/uefidump.h#L22) includes fitParser.h.
I tryed to comment this line, but it was unable to build. Please fix.

NikolajSchlej, 2016-07-16:
Fixed in the latest commit. Sorry for forgetting about it, I've fixed UEFIExtract, but just completely forgot that UEFIDump needs porting too.

closed 2016-07-16 by NikolajSchlej

laptander, 2016-07-16:
Still cannot build. Something wrong with std namespace. [Building log](https://gist.github.com/laptander/9e57e3aebfff0402e746999918f08574).

NikolajSchlej, 2016-07-16:
Can you also provide CMake log? 
I'm testing my builds in OSX with CLang (AppleClang 6.0.0.6000057), and it's totally fine for me.

NikolajSchlej, 2016-07-16:
Also, please also try the latest commit, I've added the include file where std::count and std::reverse are expected to be defined. Should add GCC to my build testing or just finally setup CI properly.

laptander, 2016-07-16:
Now it builds correctly, thanks.