LongSoft/UEFITool #60: Support for Broadwell E

Piker-Alpha, 2016-07-22:
Hi,

I ran: 
```
./UEFIPatch X99-E-ASUS-0601.CAP
```
And got this output:

parseImageFile: Aptio capsule signature may become invalid after image modifications
parseSection: section with unknown type 52h
parseFile: non-empty pad-file contents will be destroyed after volume modifications
parseSection: section with unknown type 52h
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file

Can this be fixed?

NikolajSchlej, 2016-07-23:
Hi Pike,
please try this patch:
# PpmInitialize | Broadwell-E
3FFCAE95-23CF-4967-94F5-16352F68E43B 10 P:0FBA6C24400F:0FBA7424400F

Piker-Alpha, 2016-07-23:
Done. Got this output:

parseImageFile: Aptio capsule signature may become invalid after image modifications
parseSection: section with unknown type 52h
parseFile: non-empty pad-file contents will be destroyed after volume modifications
parseSection: section with unknown type 52h
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 6 bytes at offset F69h 0FBA6C24400F -> 0FBA7424400F
Image patched

One thing that I noticed is that the patched file is 2 bytes shorter. Will that be a problem?

NikolajSchlej, 2016-07-23:
2 bytes difference could be a problem, yes. Dunno where it comes from. Can you attach or link the file?

Piker-Alpha, 2016-07-23:
Sure. Two files attached. Thank you for looking into this!

[X99-E-ASUS-0601.CAP.zip](https://github.com/LongSoft/UEFITool/files/379967/X99-E-ASUS-0601.CAP.zip)
[X99-E-ASUS-0601.CAP.patched.zip](https://github.com/LongSoft/UEFITool/files/379968/X99-E-ASUS-0601.CAP.patched.zip)

giacomoleopardo, 2016-07-25:
Pike, I see equal size from the info window of you files. Do you have some other way to check it?

Piker-Alpha, 2016-07-25:
I feel stupid. You are right. I apologise for the error. I have no idea what I did. Time to check out the patched BIOS. 

Thank you for letting me know!

Piker-Alpha, 2016-07-29:
Ok I went for it and tried to flash the patched BIOS with the Asus EZ Flash 3 Utility v03.00 (from the BIOS) and got this error: "_Selected file is not a proper BIOS_".

NikolajSchlej, 2016-08-04:
Have you tried USB BIOS Flashback? Newer ASUS boards refuse to flash modified BIOS files with standard tools, and old AFU /GAN method doesn't work either. There is a workaround for it, but it's had to perform a generic way, and UBF flashes everything, so if it's an option for that board - I would try it first.

Piker-Alpha, 2016-08-04:
I have not but I will do that today (have five new BIOS chips handy). Thanks.

NikolajSchlej, 2016-08-04:
Then I would buy some cheap CH341A-based SPI programmer like [this](http://www.ebay.com/itm/USB-Programmer-CH341A-Series-Burner-Chip-24-EEPROM-BIOS-Writer-25-SPI-Flash-NEW-/) and forget about software tools and protections.

Piker-Alpha, 2016-08-04:
Ok thanks. This is for [The big Asus X99-E plus i7-6850K give away](https://pikeralpha.wordpress.com/2016/08/01/the-big-asus-x99-e-i7-6850k-give-away/) so I'll add it to the list.

What software do I need for this?

NikolajSchlej, 2016-08-05:
Latest flashrom builds are supporting this programmer, AFAIK.

giacomoleopardo, 2016-08-05:
I’ve just flashed Asus X99-A II bios with an old version of Afudos ([download here](http://download947.mediafire.com/c4ek1bk5kazg/hx54iv6ha9vpz83/Afudos_238.zip)) with
- afudos bios.bin /o to dump the original bios
- UefiPach v. 0.3.9 with the latest broadwell-E patch (3FFCAE95-23CF-4967-94F5-16352F68E43B 10 P:0FBA6C24400F:0FBA7424400F)
- afudos bios_patched.bin /gan

I can’t check if it works because I don’t even get the OSX El Capitan 10.11.6 installer. With the intel i7 6900k cpu. Windows works good. I need some help to know how to patch the kernel, I guess. Pike’s thread is too much for my knowledge

NikolajSchlej, 2016-08-05:
/GAN method isn't reliable anymore and can lead to bricked system or just refuse to apply the modified data, so I stopped recommending it. On the other hand, HW SPI flashers are cheap and reliable.

giacomoleopardo, 2016-08-05:
Right. As a matter of fact /gan method bricked my Asrock Z10 fatal1ty ITX. I did flashed via SPI, but it allowed me to flash just the original bios from downloaded Asrock. Not the patched one

Piker-Alpha, 2016-08-05:
@NikolajSchlej, 

Ok thanks.

@giacomoleopardo,

You only need to patch the kernel the way I do if you want XCPM instead of AppleIntelCPUPowerManagement. And like Nikolaj said. Please check the lock bit with RW Everything. Thanks!

giacomoleopardo, 2016-08-05:
@NikolajSchlej 
I'm here with RW Everything opened on CPU MSR Registers. Can't recognize MSR 0xE2!

stinga11, 2016-08-05:
Someone has tried to manually patch the Broadwell-E .CAP with the UEFITool 0.21.5? because the PMPatch doesn't works.

NikolajSchlej, 2016-08-06:
Press USER button and add it manually.

giacomoleopardo, 2016-08-06:
You mean like this? What am I looking at, then?
[image]

NikolajSchlej, 2016-08-06:
Yep, bit 15 is zero, it means your 0xE2 register is unlocked now. Seems the patch works after all.

giacomoleopardo, 2016-08-06:
That's good to hear. What about this one? It's ASRock Fatal1ty Z170 ITX. I know for sure it's locked (even with SPI I couldn't flash with the unlocked one), but the bit 15 seems to be "0" either. I don't quite understand...
[image]

NikolajSchlej, 2016-08-06:
We are talking about different kind of locks here. 
If bit 15 in 0xE2 is 1, AppleIntelCPUPowerManagement.kext will crash, if it's 0 - it could work or not depending on the CPU microarchitecture. 
BIOS Lock (the ability to modify the image) is a different and unrelated story.

giacomoleopardo, 2016-08-06:
Once again thanks for clarifying. As a matter of fact in clover config.plist file, changing from
```
<key>KernelPm</key>
<true/>
```
to
```
<key>KernelPm</key>
<true/>
```
didn't compromise the boot process.
Very valuable info!

Piker-Alpha, 2016-08-20:
I can confirm that the Broadwell E patch works, but he lost the Mac address.

closed 2016-08-21 by Piker-Alpha