LongSoft/UEFITool #62: need help, Can't Patch Bios Asus X99 Strix

dedeendium, 2016-10-08:
Hi NikolajSchlej,

I'm newbie on this thing, and I'm trying to understand for patching this bios but it's showing like this: (maybe I'm wrong doing)
/UEFIPatch/patches.txt: line 30: 3FFCAE95-23CF-4967-94F5-16352F68E43B: command not found

Can you help me for patching my bios so i can get unlock MSR 0xE2 register for intel broadwell E, please?

This my bios file: 

[STRIX-X99-GAMING-ASUS-1101.CAP.zip](https://github.com/LongSoft/UEFITool/files/517899/STRIX-X99-GAMING-ASUS-1101.CAP.zip)

Thanks.

Velocet, 2016-10-09:
I don't think that GitHub is a good place to call for this type of help. Normally the _Issues_ section is for issues about the software itself.

I would suggest you make your request over at [InsanelyMac](www.insanelymac.com), [MDL](https://forums.mydigitallife.info/forums/25-BIOS-Mods), [Win-Raid](http://www.win-raid.com/forum.php) or [Bios-Mods](https://www.bios-mods.com/forum/).

dedeendium, 2016-10-09:
Thanks for you'r suggestions, i'm going to here because on the patches.txt the binary that need to find and replace it's not on that address like on the command inside the patches.txt, it's why i ask help. Will try to go to insanely mac

closed 2016-10-10 by matrosov