LongSoft/UEFITool #64: UEFIPatch not working on Gigabyte Z170.

cnrd, 2016-11-29:
Hi

I'm trying to patch MSR 0xE2 on the newest BIOS for GA-Z170X-UD3 (F20), but UEFIPatch is unable to patch it.

`./UEFIPatch Z170XUD3.F20
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file`

It did work correctly with eariler versions of the BIOS, so I'm thinking that the problem must be the addition of KabyLake support?

I tried extracting the SiInit TE Image section and searching for 75080D00800000 using a hex editor, but the hex string is no where to be found in that.

I think we need a new patch for KabyLake / updated SkyLake boards?

The BIOS can be found here if that is of any help: http://www.gigabyte.com/products/product-page.aspx?pid=5499#bios

Hoping that you can shed some light on this. :-)

Piker-Alpha, 2017-01-25:
Can you try this Gigabyte F21 BIOS patch:

# SiInit | Kaby Lake
299D6F8B-2EC9-4E40-9EC6-DDAA7EBF5FD9 10 P:81E10080000033C1:9090909090909090
299D6F8B-2EC9-4E40-9EC6-DDAA7EBF5FD9 12 P:81E10080000033C1:9090909090909090

Herrnobiz, 2017-02-03:
Different board, a Gigabyte ga-z170-wifi with bios F20, same problem. I even tried those new patches from piker-alpha but same results

NikolajSchlej, 2018-02-15:
Please try now, should work.

closed 2018-05-07 by vit9696