LongSoft/UEFITool #67: Newer UEFITool unable to modify image

shuffle2, 2016-12-17:
I have some image which I am able to successfully replace a PE file in on older UEFITool.
On current UEFITool, the options to allow modifying the image are always grayed out. I haven't looked into why, but maybe it is because the FIT claims it supports Boot Guard. In any case, UEFITool should not prevent changing this image.

Cr4sh, 2016-12-22:
Unfortunately, NE (aka new engine) brach of UEFITool currently does not support image editing at all. If you interested in this functionality -- try to use older version, for example, this one: https://github.com/LongSoft/UEFITool/releases/tag/0.21.5

carlosmogo, 2016-12-29:
Hi, I have an asus X99A and a processor e5 2630 v4 works fine for the first time but after restart it appears Q code 19, please you can patch the bios is this
http://dlcdnet.asus.com/pub/ASUS/mb/LGA2011/X99-A/X99-A-ASUS-3402.zip?

Magendanz, 2017-03-11:
So is image editing something that will be coming soon in NE builds?  This seems like core functionality and the menu commands still remain.

lasalvavida, 2017-08-18:
It would be nice to be able to use some of the new engine features, but image editing is a must. Is there any way we could get a series of issues outlining specific things that are still to do in the new engine with some direction on what needs to be done? For example in [ffsbuilder](https://github.com/LongSoft/UEFITool/blob/new_engine/common/ffsbuilder.cpp), making issues for methods that return U_NOT_IMPLEMENTED along with what the method needs to be able to do in order for it to be considered to be implemented.

The reason I say this is because if I have spare cycles, I'd be happy to help where I can, but without a specific task/requirements it would be difficult for someone outside the development loop to contribute.

osresearch, 2017-11-03:
Adding a vote for image editing in the `new_engine` branch.  I'm working with UEFI images that use FFSv3. The old engine tool can't parse this version of the file system structure, so I can't replace individual DXE and PEI files.

interferenc, 2017-11-30:
Another vote for ffsv3 support in EUFIPatch, please.

osresearch, 2017-11-30:
I've modified the old engine to be able to regenerate FFSv3: https://github.com/LongSoft/UEFITool/pull/90

it "works for me" on the Intel firmware images on the s2600wf mainboard that I'm testing with the NERF branch of heads: https://github.com/osresearch/heads/tree/nerf

SourceCodeDeleted, 2020-01-13:
Is this still a problem? I can check and try tonight if needed.

NikolajSchlej, 2022-09-13:
I will start working on this huge task when I'm done with smaller parsing-related ones. Not many of them left though, so I hope to deliver before the end of this year.

jelischer, 2022-11-21:
Please update the README as to whether the NE version can now edit images. There are  comments in the wild about how you need to use the "non-NE" version to be able to edit but no comments here as to whether that is still true or when it was changed, or in fact as this bug suggests, there are limitations to the editing ability of the NE version.  It makes it hard to work out what to download and use.

wravoc, 2023-12-07:
Since the LogoFAIL vulnerability has been discovered, this may be more of a priority.

htm23x, 2024-06-10:
Any news/dev on NE supporting bios mod/writting?
We really need this feature being implemented, please.
Thanks.

Jan64X, 2024-09-27:
+1, pretty important to implement.
I'm trying to patch an image, so guess I have to try the older version...

Bill50han, 2024-10-14:
+1, I try to use uefitool 0.28 to replace an EFI module, but it warnd "invalid volumn". I think I need NE version to edit it.

NikolajSchlej, 2025-02-25:
I'm back to the land of funemployment for a while, and this main feature will be prioritized accordingly now. 

Currently I'm working on rewriting NVRAM parsers into KaitaiStruct to ensure that they can withstand fuzzing, next will be creation of a test suite and incorporation of automated testing into CI/CD, next will be the new builder code.

There's no way I will be able to write a working builder without a comprehensive test suite, given the amount of images that need to work after rebuild and the amount of quirks found in the wild since the last builder had been written in 2013.

Forkinator, 2026-05-14:
Just casting my vote for this. Really looking forward to being able to modify images. Thanks to everyone who worked on this project.

Privex-chat, 2026-10-14:
still hoping