LongSoft/UEFITool #70: Way to work in bulk

mirh, 2017-02-20:
My use case is: I dunno what I'm looking for, so I want to extract all PE32 images, or compressed sections to then look into them with IFR extractor.

Putting aside this would be cool to script in a CLI or something (I see you already have plenty of people asking for that).. Why CTRL+mouse click doesn't work?
It's a pain having to go through all the right clicking every time.

Cr4sh, 2017-03-09:
You can use UEFIExtract tool for batch extracting of all of the firmware image contents.

mirh, 2017-03-09:
Uh, that's really really really neat indeed. 
Though.. I still feel like you _could_ improve UEFITool GUI?

AdrianKoshka, 2017-03-09:
just *my* two cents, I'd assume people doing batch loads want a nice automated scripted environment to do it in, not a GUI.

mirh, 2017-03-09:
Better?

NikolajSchlej, 2020-01-25:
I've re-evaluated the amount of work to be done here and decided not to proceed with this request, because the UI will need to be basically rewritten from scratch.

closed 2020-01-25 by NikolajSchlej