LongSoft/UEFITool #72: Can't open RSA signed Insyde image

mirh, 2017-03-03:
`UEFI volumes not found` error is returned. 

Offender can be found [here](http://www.mediafire.com/download/9i4lvtpe72klu6r/sp55068.zip).

Cr4sh, 2017-03-09:
Hi,
UEFITool was designed to work mostly with firmware images stored on the flash chip, it might not support some vendor-specific containers with firmware updates (like your one, for example) because it's total pain to implement it. 
The right way to process this image with UEFITool -- flash it to your motherboard using vendor supplied firmware update software, dump flash chip contents (using CHIPSEC or hardware programmer) and load obtained dump into the UEFITool like regular UEFI flash image.

Cr4sh, 2017-03-09:
Also, you can try to use this utility to pre-process Insyde image: https://www.dropbox.com/s/hf6y2derwyh9mwy/hewprsa.exe?dl=0

mirh, 2017-03-09:
Yep, that works (and I guess I should have mentioned that anyway I had it working after passing through Phoenix Mod Tool). 

I also guess it's your call ultimately if you want to support it one day (or at least tell/warn the user about the thing) or not.

closed 2017-10-19 by matrosov

0xfede7c8, 2020-10-01:
@Cr4sh Do you know where can I get the source code for that tool?