LongSoft/UEFITool #78: UEFITool_0.21.5_win - modified and rebuilt Dell BIOS does not boot properly

sun2sirius, 2017-05-23:
Just wanted to drop a note that both the NE and 0.21 tools work very well to extract EFI components. The older tool also modified the BIOS image I experimented with - I needed to remove one UEFI Application. The resulting BIOS image looked great when re-opened with the tool, but unfortunately the image did not boot on the device. It got into BIOS recovery mode and I had to recover the original BIOS. I do not know whether this is expected or maybe there is some trick to it. Or maybe the image was rebuilt OK, but it verifies signing (can it be signed?) and refuses to boot.

As a side note, UEFITool_0.21.5_win.zip cannot download correctly as it is blocked by antivirus software. Chrome says "Failed - Virus detected".

aphroteus, 2017-06-01:
I follow the steps, BIOS still boot on the device, no matter "Secure Boot" is enabled or not.

1. Remove an .efi application
2. Rebuild (save image file...)

Or
1. Replace a PE32 image section
2. Rebuild (save image file...)

The experiment is interesting for me, I was wondering where is the BIOS stuck at in your case?
Is it possible failed to locate firmware volume of efi in Dxe phase? And just stuck there...

vutung2311, 2017-10-20:
As far as I know it's about fail signing key checking. Dell has enable Intel Boot Guard so you can't modify and flash it and hope it work correctly because you don't have the private key to sign the image. However there some part of BIOS image that you can't change and it will still pass signing key check like the NVRAM part. You can use NE v43 to see the part of BIOS image that you can change.

eeff11, 2017-10-20:
RSA signed,wondering about no luck in brute force hacking with dictionaries.

NikolajSchlej, 2017-12-03:
BootGuard is the reason, UEFITool can do nothing to prevent it.

closed 2017-12-03 by NikolajSchlej