LongSoft/UEFITool #81: UEFITool Parse Problem With Asus AMI Aptio V 6Mb Bios.

Zilog80c, 2017-08-27:
Hi to,
 all; when parse this bios not appers region of bios Me, Descriptor, ecc...
[image]

I Search of extract Me firmware, but i don't find.
Thaks a lot.

Zilog.

skochinsky, 2017-08-27:
It seems this contains only the BIOS region; to get the ME you'll probably need to dump the SPI chip with an external reader.

Zilog80c, 2017-08-28:
....mmm....i don't thinks, becasue in the site of vendor there is explit write that: 
Update NV VBIOS
Update Intel ME firmware

That there is Intel Me Firmware.

Thanks a lot.
Zilog.

skochinsky, 2017-08-28:
got a link to the update?

Zilog80c, 2017-08-28:
Directly from support page site of Asus.

skochinsky, 2017-08-28:
well, what is the link then?

Zilog80c, 2017-08-28:
https://www.asus.com/Laptops/VivoBook-Pro-N552VW/HelpDesk_Download/

Select Windows 10
Section Bios. 
Version 202

skochinsky, 2017-08-28:
Thanks for the link.

I could not find any sign of the ME firmware in that update. I suggest you to ask folks at [Win-Raid](http://www.win-raid.com/f39-Special-Intel-Management-Engine.html), maybe they can figure out where it is hidden.

platomav, 2017-08-29:
At most laptops, ASUS gives BIOS/EC region only updates which are encapsulated. So no FD, ME, GbE etc regions. So it is as Igor said [above](https://github.com/LongSoft/UEFITool/issues/81#issuecomment-325220430).

skochinsky, 2017-08-30:
@platomav  so the claim "Update Intel ME firmware" is false?

platomav, 2017-08-30:
@skochinsky Technically yes, practically no. It's a matter of bad wording/translation. They could mean that the BIOS modules which communicate with the ME got updated. ASUS tends to release some "ME Update Tool" packages, when laptops need a ME firmware update, which just use FWUpdate tool. It could also be random changelogs as most of ASUS's always say "Improve system stability" even if they incorporate very big changes. It doesn't really matter, one thing is certain: there is no ME firmware inside those encapsulated BIOS/EC updates.

closed 2017-10-19 by matrosov