LongSoft/UEFITool #83: official A43 build does not run due to missing Qt5 libs

skochinsky, 2017-10-12:
Running the latest UEFITool_NE_A43.exe produces such messages:

---------------------------
UEFITool_NE_A43.exe - System Error
---------------------------
The code execution cannot proceed because Qt5Widgets.dll was not found. Reinstalling the program may fix this problem. 
---------------------------
OK   
---------------------------

So it seems you now link to Qt libs dynamically, however this dependency is not mentioned anywhere on download page. TBH I would prefer the previous behavior of single exe with no extra dependencies.

carlosgbr, 2017-10-16:
have the same problem on a Windows 10 Pro system Build 10586.962

nexus76, 2017-10-18:
Microsoft Windows [Version 10.0.16299.19]
I'm getting a 0xc000007b pop-up, just doesn't launch, even though I copied 

Qt5Core.dll
Qt5Gui.dll
Qt5Widgets.dll

into the same directory

vutung2311, 2017-10-18:
I managed to compile a working version using this guide:
https://wiki.qt.io/Building_a_static_Qt_for_Windows_using_MinGW
I leave my compiled one here in case any of you need to use it.
https://drive.google.com/file/d/0B_-DWvOWTmeZZVhIUnBJSkFFNGc/view?usp=sharing

matrosov, 2017-10-19:
fixed

closed 2017-10-19 by matrosov

walkman8196, 2017-11-06:
NE_A44 the same problem

matrosov, 2017-11-06:
fixed

matrosov, 2017-11-06:
> how is it fixed? if I open version 44 and this bug is present.

Before making the comments like this double check you are using a recent
version of the binary. I updated the release on the morning today ;) Thank
you for the feedback!

walkman8196, 2017-11-07:
Thanks, now works  👍