LongSoft/UEFITool #87: How to get UEFIReplace working?

KingBonecrusher, 2017-11-25:
I have a full backup from an eeprom containing everything. This works well with UEFITool. I know what i want to replace, but i will do this in a script with no user interaction. So i hope UEFIReplace is the right way. The section is inside the image under "BIOS Region" -> "NVRAM" -> "VSS Store" -> "TO REPLACE"

I have used the GUID from the "TO REPLACE" section, but i get everytime a INVALID PARAMETER error from UEFIReplace. Is there any readme or example how to use it correctly?

NikolajSchlej, 2017-11-27:
UEFIReplace is based on master, while UEFIEtract is based on new_engine. Master doesn't have NVRAM parser, so you can't replace a VSS store using it, only the whole firmware volume.

KingBonecrusher, 2017-11-27:
Thank you for the fast reply! Maybe a task for the future ;-) I`am using new_engine and also master branch for my rom collection. Very happy with UEFITool! I have written a small bash script which parses the firmware and make the changes, my solution for now.

closed 2017-12-03 by NikolajSchlej