LongSoft/UEFITool #89: Intel Flash Descriptor v2 Regions

platomav, 2017-11-27:
For Intel FD v2, these are the names of all regions in order:

Flash Descriptor
BIOS
Management Engine
Gigabit Ethernet
Platform Data
Device Expansion 1
Secondary BIOS
CPU Microcode
Embedded Controller
Device Expansion 2
Innovation Engine
10 Gigabit Ethernet 1
10 Gigabit Ethernet 2
Reserved 1
Reserved 2
Platform Trust Technology

Can you add parsing for them with the proper names as shown above?

skochinsky, 2017-12-07:
how do you detect descriptor v2?

platomav, 2017-12-07:
There is no actual indication from Intel but FLCOMP > Read Clock Frequency has proven to work just fine so far. It is always 20MHz for FD v1 and 17MHz for FD v2.

NikolajSchlej, 2017-12-10:
Any sample files with those?

platomav, 2017-12-10:
Any Skylake+ SPI uses FD v2. All Apollo Lake SPI use Device Expansion 1 (called Reserved 1 at UT). I recently found a server SPI image from HP which had firmware at the Innovation Engine. I don't have the link anymore but I'll look. As for the order/naming, it is correct based on spsFIT v4. One strange thing is that the last partition is called Reserved 3 at one place and PTT at another. I went with PTT.

matrosov, 2017-12-11:
fixed in A45

closed 2017-12-11 by matrosov