LongSoft/UEFITool #91: Support for Skylake-X

interferenc, 2017-11-28:
Hey guys,

If I try to patch my bios with UEFIPatch, the KabyLake patch gets applied. Could be because of KabyLake-X support on my motherboard, I don't know.

What I do know, is that after flashing the bios, MSR 0xE2 is still locked. SO whatever the KabyLake patch did, it did not unlock my 0xE2. My guess would be that an x299 path will be closer to the x99 broadwell-e patch.

Any idea how to find a new patch? Sadly, reverse engineering UEFI is way above my head.

Any help is appreciated.

Thank you.

M

danielstuart14, 2017-11-28:
After some discussion [here](https://github.com/danielstuart14/ASUS_BIOS_REPO/issues/12#issuecomment-347569196), I think we found the issue.

It seems that x299 has an Init for both generations of processors (Kabylake-x and Skylake-x), as you can see in this image:
[image] (x299 in the left and z270 in the right)

So, I propose this patch:
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 10 P:81E10080000033C1:9090909090909090
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 12 P:81E10080000033C1:9090909090909090

I'm still waiting for @interferenc test it, so a PR can be made.

NikolajSchlej, 2018-02-15:
New patches integrated.

closed 2018-02-15 by NikolajSchlej

chriz74x, 2019-04-13:
I tried uefipatch with skylake bios from HP g400 desktop and it doesn’t find anything to patch. I opened an issue but it was closed without any help or further explanation. Can anyone at least point me to what should I check for in the extracted bios ?

CrazyHulk, 2019-12-24:
> I tried uefipatch with skylake bios from HP g400 desktop and it doesn’t find anything to patch. I opened an issue but it was closed without any help or further explanation. Can anyone at least point me to what should I check for in the extracted bios ?

Hi.
Can u tell me how to patch?

NikolajSchlej, 2019-12-24:
> HP g400
Link to BIOS file?

CrazyHulk, 2019-12-24:
I have similar problem.
[ftp://asrock.cn/BIOS/Server/EPC621D8A(2.10)ROM.zip](Bios file)

I try to disable CFG Lock with no lucky.
I found some CFG Lock labels in bios image.(0xB3, 0xB4, 0xB5, 0xB6)

For example:

```
0x59B2C         One Of: SAPMCTL_CFG LOCK, VarStoreInfo (VarOffset/VarName): 0xB3, VarStore: 0x5, QuestionId: 0x1835, Size: 1, Min: 0x0, Max 0x1, Step: 0x0 {05 91 DC 19 E4 19 35 18 05 00 B3 00 10 10 00 01 00}

0x59B3D             One Of Option: Disable, Value (8 bit): 0x0 {09 07 B5 00 00 00 00}

0x59B44             One Of Option: Enable, Value (8 bit): 0x1 (default) {09 07 90 0D 30 00 01}

0x59B4B         End One Of {29 02}
```

So, I get variable '0xB3' and value '0x5', but option is '0x0' or '0x1'.

'setup_var_3 0xB3' in GRUB shell get 0x06.
'setup_var_3 0xB4' in GRUB shell get 0x07.
'setup_var_3 0xB5' in GRUB shell get 0x08.
'setup_var_3 0xB6' in GRUB shell get 0x09.
I try to set all variable to 0x00, still in lock mode.

And I had try patch pattern.
UEFIPatch version 0.26.0.
```
INPUT:
./UEFIPatch C621D8A2.10 patches.txt -o patched

OUTPUT:
parseFile: non-empty pad-file contents will be destroyed after volume modifications
No patches can be applied to input file
```
It seems nothing happened.

NikolajSchlej, 2019-12-24:
SAPMCTL_CFG has nothing to do with the register you want to lock, this one is about various bits is MSR 0x1FC, and the register you want to unlock is 0xE2.

NikolajSchlej, 2019-12-24:
@chriz74x, I could not find a place in your BIOS backup (downloaded from the closed issue you've mentioned) that sets a lock on 0xE2 register. Are you sure that the the register is locked for you? Can you show RW Everything window for 0xE2 MSR with bit 15 set?

CrazyHulk, 2019-12-25:
> SAPMCTL_CFG has nothing to do with the register you want to lock, this one is about various bits is MSR 0x1FC, and the register you want to unlock is 0xE2.

Thx for ur reply.
If there no CFG Lock found (All I found is SAPMCTL_CFG), did it means that I can not unlock 0xE2?

NikolajSchlej, 2019-12-25:
I probably means that there's no option to unlock it from BIOS Setup, so a patch is required. I'll see if I can find a place to patch in your BIOS later this week.

CrazyHulk, 2019-12-26:
> It most likely means that there's no option to unlock it from BIOS Setup, so a patch is required. I'll see if I can find a place to patch in your BIOS later this week.

Thx.

[image]

Do u need more information about my bios?

NikolajSchlej, 2019-12-26:
No, will check it out once I have some time for it.

CrazyHulk, 2020-01-09:
Is here any progress?

NikolajSchlej, 2020-01-14:
Got sick, so there's none yet. Will try again when I get better.

NikolajSchlej, 2020-01-14:
@CrazyHulk, your BIOS is fully covered by Intel BootGuard, and might refuse to boot if it's enabled for real. You can check of BootGuard is enabled on your system by reading MSR 0x13A. If a bit #32 (0x100000000) is set, BG is active and 3rd-party BIOS modifications aren't possible on your motherboard.

CrazyHulk, 2020-01-15:
I am sorry to hear that u got sick. Hope u will be better soon.

I have check MSR 0x13A.
  
[image]

It seems that no BootGuard here.

NikolajSchlej, 2020-01-15:
Thanks. Bit 34 = 1 (aka register value 0x400000000) is BOOTGUARD_DISABLED, good news. I'll try to find the place to patch once I get better.

NikolajSchlej, 2020-01-16:
@CrazyHulk, try this mod first (flash it using SPI programmer if normal ways of flashing it fail) and report if it boots or not. It has DXE verification code disabled and DXE volume modified (useless PEI files removed from there), so if it works - it will be base for all further modifications for this platform.
[C621D8A210.zip](https://github.com/LongSoft/UEFITool/files/4070536/C621D8A210.zip)

CrazyHulk, 2020-01-22:
@NikolajSchlej Hi, I am sorry to reply so late.
I have flash it with instant flash(bios support this feature), it seems that everything goes well.

NikolajSchlej, 2020-01-25:
@CrazyHulk, please test the attached image. The patch is very similar to the current Skylake-X patch for CpuInitPei, but uses a different image type (TE instead of PE) and different register (ebx) to hold the OR value.

New patch string would be as follows:
> # CpuInitPei | Skylake-X
> 01359D99-9446-456D-ADA4-50A711C03ADA 12 P:B.0080000023C.0B:B.0000000023C.0B 
> 01359D99-9446-456D-ADA4-50A711C03ADA 10 P:B.0080000023C.0B:B.0000000023C.0B 

However, your image is special because it has 10GBE region that is unknown to the old engine UEFIPatch is based on, so before that descriptor issue is resolved, usage of UEFIPatch on your BIOS image will break it, so I made a patch manually.
[C621D8A210.mod2.zip](https://github.com/LongSoft/UEFITool/files/4112217/C621D8A210.mod2.zip)

CrazyHulk, 2020-03-04:
Sorry to reply to your message so late. 

Thx for ur job.
I have test image (C621D8A210.mod2.zip), it seems that everything goes well.

CrazyHulk, 2020-03-04:
> @CrazyHulk, please test the attached image. The patch is very similar to the current Skylake-X patch for CpuInitPei, but uses a different image type (TE instead of PE) and different register (ebx) [...]

Sorry to reply to your message so late.

Thx for ur job.
I have test image (C621D8A210.mod2.zip), it seems that everything goes well.

CrazyHulk, 2020-03-18:
@NikolajSchlej 
Any suggestion here?

NikolajSchlej, 2020-03-18:
Will check again on the weekend.

CrazyHulk, 2020-03-27:
@NikolajSchlej 
I am glad to hear that.
Waiting for your good news.

Thank you.

CrazyHulk, 2020-05-04:
[image]
It was weird， I found that  0xE2 MSR with bit 15 set became 0 today, but it to 1 again after reboot.
It seems that it works sometimes...
But I do not know how to make it always works, I just make a normal boot today.

CrazyHulk, 2020-05-04:
@NikolajSchlej 
It works (0xE2 MSR with bit 15 set become 0) after wake from sleep...