LongSoft/UEFITool #92: UEFIPatch complains about

interferenc, 2017-11-29:
When patching a GUID twice, I get the error message:

```
reconstructVolume: root volume can't be grown
```

But if I patch it in two rounds, all is well. Is this normal? Shouldn't I be able to patch a BIOS with these patches in one round?

```
# Skylake-X
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 10 P:81E10080000033C1:9090909090909090 
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 12 P:81E10080000033C1:9090909090909090 
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 10 P:0D008000008A8E:25FF7FFFFF8A8E 
D71C8BA4-4AF2-4D0D-B1BA-F2409F0C20D3 12 P:0D008000008A8E:25FF7FFFFF8A8E 
```

Thanks.

NikolajSchlej, 2017-12-03:
Link to the image file?

interferenc, 2017-12-04:
http://dlcdnet.asus.com/pub/ASUS/mb/LGA2066/ROG_STRIX_X299-E_GAMING/ROG-STRIX-X299-E-GAMING-ASUS-1004.zip

NikolajSchlej, 2018-02-15:
Should work now, please verify.

interferenc, 2018-02-15:
I am sorry to report that for me at least it still does not work:

```
octa:UEFIPatch interferenc$ ./UEFIPatch ROG-STRIX-X299-E-GAMING-ASUS-1102.CAP
parseImageFile: Aptio capsule signature may become invalid after image modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 8 bytes at offset 2CB7Ch 81E10080000033C1 -> 9090909090909090
patch: replaced 8 bytes at offset 2CB7Ch 81E10080000033C1 -> 9090909090909090
patch: replaced 7 bytes at offset 2CB64h 0D008000008A8E -> 25FF7FFFFF8A8E
patch: replaced 7 bytes at offset 2CB64h 0D008000008A8E -> 25FF7FFFFF8A8E
reconstructVolume: root volume can't be grown
Error
```

vit9696, 2018-05-07:
Ok, I fixed it up in master :)

commit ab8851fe (2018-05-07): Do not try to rebase removed sections, fixes #92
closed 2018-05-07 by vit9696