LongSoft/UEFITool #93: NVRAM parsing hp elitebook 840 g1

KingBonecrusher, 2017-11-29:
Hi! 
I have an bios rom from an hp 840 g4. The nvram could not be parsed. It looks like hp uses AA553F instead of A557F for the start. Could i send you this file maybe for parsing?

NikolajSchlej, 2017-12-03:
Link to the image file?

KingBonecrusher, 2017-12-03:
https://ufile.io/drt4i

NikolajSchlej, 2017-12-10:
What version are you using? Can't reproduce on A44, parsed OK.
[image]

closed 2017-12-10 by NikolajSchlej

KingBonecrusher, 2017-12-11:
You`re correct. It was my fault. I`ve used alpha40. The alpha44 was compiled as shared library and i haven`t used it. Can i compile it as non shared executable, or is this because of position independent executable?