LongSoft/UEFITool #94: How to get EC firmware from BIOS?

fractalawareness, 2017-11-30:
I have broken Vaio Duo 13 SVD132A14L with no warranty. Local repair guys have replaced broken EC, but can not find firmware anywhere (it's kind of a rare device). Can anybody please help me extracting EC firmware from BIOS or BIOS dump?
I've extracted the BIOS rom file from the [VAIO BIOS update utility](http://dlv.update.sony.net/US/pc/ISBSYS-P0322046-1102.EXE). Also I've found a [BIOS dump from a real device online](https://yadi.sk/d/_b9z184o3GzsXx). But couldn't extract the EC firmware from these files because the lack of knowledge how exactly to get the EC firmware addresses etc.
Exuse me if it's the wrong place, I've been asking everywhere and nobody can help. I'm just trying to help the repair guys that have been trying to fix my laptop for almost a year already.
Thanks!

platomav, 2017-11-30:
I was able to extract the BIOS region (not full SPI, use UEFITool to replace on your/3rd-party dump) and EC (what you want) from the VAIO BIOS Update Utility. Note that the EC is not found inside the SPI image dump so it needs to be flashed on the EC chip (1Mbit or 128KB size), wherever that is.

[ISBSYS-P0322046-1102_R1082S7.zip](https://github.com/LongSoft/UEFITool/files/1517730/ISBSYS-P0322046-1102_R1082S7.zip)

fractalawareness, 2017-11-30:
Thank you sooo much!
I've forwarded this file to the repair guys immediately 🚀

timofonic, 2017-12-01:
@fractalawareness I wonder how it's made in your laptop. Mine uses the same EC chip used by OLPC. There are tools, but not Open Source.

Good to see activiity here. I hope some people relaunch this amazing project and makes it official...

imranromi, 2017-12-03:
SOME LENOVO LAPTOP EC BIOS (128KB) INSIDE THE MAIN BIOS.

NikolajSchlej, 2017-12-03:
Unrelated to UEFITool, closing.

closed 2017-12-03 by NikolajSchlej