LongSoft/UEFITool #95: DELL LATITUDE BIOS EXTRACT.

imranromi, 2017-12-03:


imranromi, 2017-12-03:
HI EVERYONE I AM NEW THIS FORUM I NEED SOME HELP HOW TO EXTRACT BIOS DELL LATITUDE EXTRACT TO ROM/BIN/HEX ANYBODY KNOW PLZ SHARE THANX IN ADVANCE.

NikolajSchlej, 2017-12-03:
This is not a forum, there's no need to write in all caps, you don't have a link to your BIOS, UEFITool is not a tool to extract OEM formats.

closed 2017-12-03 by NikolajSchlej

imranromi, 2017-12-04:
Oh Sorry thanx for reply i mistake but i need how to extract dell latitude
series  bios extract bin/hex/rom file can you help me.
Here is link bios.
https://downloads.dell.com/FOLDER04348801M/1/E6440A18.exe

goodwin, 2017-12-04:
It will not help you too much if you will open stock image. You will not be able to write it properly (i think so). You have to dump your current UEFI image from running machine - it is possible on Dell machines with Universal BIOS Backup ToolKit

imranromi, 2017-12-04:
Thanx for reply i try this software not work proper the main bios is 8mb this software show 6.sum.