LongSoft/UEFITool #96: Code optimization/avoiding possible problem suggestion

starter48, 2017-12-07:
I've found the following ugly statements in the code:
`alignment = (UINT32)pow(2.0, alignmentPower);`
Such statement are relatively slow (the power to be computed by FPU) and unreliable on some FPU implementations due to truncation error of inexact value. For example, pow() function can return inexact value 12345677.99999 instead of 12345678.0, and this value will be truncated to 12345677 instead of 12345678 respectively.
I suggest to change such statements to pure integer math:
`alignment = 1UL << alignmentPower;`
I.e. just shift **1** left by power exponent.
Note: here **alignmentPower** must be an integer in range [0..31] (you may use **1ULL** if not enough). I see the exponent argument always satisfy this requirement in this project.

matrosov, 2017-12-11:
fixed in A45 and 0.22.0

closed 2017-12-11 by matrosov

starter48, 2017-12-11:
Yet another line in ffsengine.cpp:
`alignment = (UINT32)pow(2.0, (int)(volumeHeader->Attributes & EFI_FVB2_ALIGNMENT) >> 16);`