LongSoft/UEFITool #97: FfsFile->State should be inverted when a file is saved "as-is"

osresearch, 2017-12-10:
When Ffs files are saved "As-Is", the bits are written exactly as they are stored in the firmware image.  Unfortunately that makes them unusable* with the edk2 tool `GenFv` since it expects to invert the `FfsFile->State` based on the firmware volume's erase polarity, which is almost always 1:

```
if (FvHeader->Attributes & EFI_FVB2_ERASE_POLARITY) {
    FfsFile->State = (UINT8)~(FfsFile->State);
}
```

@NikolajSchlej pointed out that this is really a bug in edk2 since a) the `FfsFile->State` has a reserved bit 0x80 that must be 0 so edk2's `GenFv` can determine if it needs to invert the state, and b) defaulting to `EP=0` for the output of `GenFfs` and then inverting it in `GenFv` is suboptimal since *no one uses `EP=0`*.

However, it would be nice if `UEFITool` would undo the inversion on export so that new firmware volumes can be constructed with `GenFv`.

___
*: Unusable in this case means that the file will be ignored when `DxeCore` discovers drivers since it thinks the state is inactive.

NikolajSchlej, 2017-12-10:
I don't think it's an UEFITool bug, but a GenFv bug instead. 
"Extract as is..." should mean it - as is, no changes. Also, 0xF8 state is what any other tool expects an FFS file to be (I tried AMIBCP, EZH2O and PhoenixTool), so the only tool that doesn't behave is GenFv, that's opensource and can be easily patched to something like this:
```c
#define EFI_FFS_FILE_ERASE_POLARITY 0x80 // Defined as "all reserved bits should be set to volume's ERASE_POLARITY" in UEFI PI spec
if ((FvHeader->Attributes & EFI_FVB2_ERASE_POLARITY) 
    && !(FfsFile->State & EFI_FFS_FILE_ERASE_POLARITY)) {
    FfsFile->State = (UINT8)~(FfsFile->State);
}
```

osresearch, 2017-12-10:
Yeah, it is definitely an edk2 issue, not a UEFITool one.  I wanted to document here so that anyone else (all six people who seem to do EFI firmware hacking) who ran into it knew why their dxe drivers were being skipped...

For the Heads / LinuxBoot case we can apply that patch to edk2 as part of our build.  That's a good work-around, thanks!

NikolajSchlej, 2017-12-10:
One other thing is that GenFv should disregard the original value of `FfsFile->State` completely, because as a volume generator it should ensure that all files in the volume are in properly constructed state:
```c
UINT8 state = EFI_FILE_DATA_VALID | EFI_FILE_HEADER_VALID | EFI_FILE_HEADER_CONSTRUCTION;
if (FvHeader->Attributes & EFI_FVB2_ERASE_POLARITY)
    state = ~state;
FfsFile->State = state;
```

vit9696, 2018-11-12:
I believe the bugreport should be filed at edk2-devel, not like we can do anything on our end.

closed 2018-11-12 by vit9696