LongSoft/UEFITool #100: change the bios volume size

tonno2, 2017-12-22:
I managed to put my 64mb bios on a 128mb eeprom I would need to change the size of a volume, is it possible?
Thank you
translated with google I do not speak English

skochinsky, 2017-12-22:
You will  probably need to edit the flash descriptor so that it has correct pointers to the regions (ME/GbE/BIOS). Use the [ifdtool](https://review.coreboot.org/cgit/coreboot.git/tree/util/ifdtool) to manipulate it. Note that you will probably need to update the VSCC table and other flash chip-related parameters (not supported by ifdtool).

tonno2, 2017-12-22:
thanks, but this program is for linux or with what program I can use it is not for windows.
thank you

NikolajSchlej, 2017-12-22:
It's possible but difficult and the success of the operation depends heavily on firmware vendor and version. [Here](https://forums.mydigitallife.net/threads/can-i-use-a-larger-capacity-bios-chip-than-the-original-that-came-with-my-motherboard.33818/#post-866356) I've answered a similar question long ago, that answer is still true for now.

tonno2, 2017-12-27:
But you can not add this option in Uefitool. Thanks

closed 2018-05-06 by vit9696