LongSoft/UEFITool #103: not work with ASRock and Gigabyte X299 BIOS

BJY1991, 2018-01-11:
i tried ues uefipatch to unlock MSR with asrok and gigabyte X299 bios but not work

parseSection: decompression failed with error "Standard decompression failed"
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 8 bytes at offset 34A5h 81E10080000033C1 -> 9090909090909090
patch: replaced 8 bytes at offset 28855h 81E10080000033C1 -> 9090909090909090
reconstructSection: executable section rebase failed
Error ,

fabiosun, 2018-01-13:
I have tried and I have this output for both of them
I am using uefipatch and patches.txt you can find in uefipatch home
I don't know if it is patched or not but it creates patched files

Last login: Sat Jan 13 13:53:57 on ttys000
fabios-Mac-Pro:~ fabio$ cd desktop
fabios-Mac-Pro:desktop fabio$ /Users/fabio/Desktop/UEFIPatch X29IT1.03
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 8 bytes at offset 4125h 81E10080000033C1 -> 9090909090909090
Image patched
fabios-Mac-Pro:desktop fabio$ /Users/fabio/Desktop/UEFIPatch X299UD4.F4
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 8 bytes at offset 34A5h 81E10080000033C1 -> 9090909090909090
Image patched
fabios-Mac-Pro:desktop fabio$

BJY1991, 2018-01-13:
my bad
i use another patches.txt which include SkyLaks-X and  KabyLake-X

NikolajSchlej, 2018-02-15:
Please verify, should work now.

closed 2018-05-06 by vit9696