LongSoft/UEFITool #105: Bad Image Relocation Entry

davidm71, 2018-01-21:
Everytime I try to replace the Microcode file GUID 17088572-377F-44EF-8F4E-B09FFF46A070 body on an MSI X299 SLI Plus MB I get an error when I try to replace the body and save the file on UefiTool Ver 22.1.

Thanks

serg-pushkarev, 2018-01-28:
Hello, @davidm71. I corrected the error and created a pull request (you can see it :)), but my changes are not accepted. You can use my fork version: https://github.com/serg-pushkarev/UEFITool.

davidm71, 2018-01-28:
Ok Thank you. I suppose I have to compile this myself?

Thanks

serg-pushkarev, 2018-01-29:
@davidm71, precompiled version https://yadi.sk/d/N_SJA7em3RtkHD. You need to perform some actions for the final firmware to work. You have to expect, what final address of PEI_CORE is correct. You need to open the firmware in the hex editor and make sure that the address is set correctly or correct it. (See pics 1 and 2: https://yadi.sk/a/S2Ix1Soy3RtnCn). Good luck! :)

davidm71, 2018-01-30:
Thank you for the link and instructions however I do not understand this part about PEI_CORE. Are you saying changing size of the Microcode file affects PEI_CORE addressing? If so then you have to go to offset FFE0h to find the table that holds the address values? Is this something that needs to be done everytime for every bios file because I never did that before and encountered a problem. Sorry for all the questions. David.

davidm71, 2018-01-30:
Also I noticed that you did not use a static QT Compiler. No problem anyhow...

davidm71, 2018-01-30:
Got some errors in the comment window. Not sure if relevant:

parseFile: invalid header checksum C1h, should be 4Bh
parseFile: invalid data checksum 11h, should be 16h
parseFile: unknown file type 22h
parseVolume: unaligned file F1E4874C-90BF-AF1D-9743-BAA643538B0B

davidm71, 2018-01-30:
Its like every thing past 17088572-377F-44EF-8F4E-B09FFF46A070 on the last volume is no longer there:

https://www.dropbox.com/s/57pt679arc7azx3/uefitool_bug.png?raw=1

davidm71, 2018-01-30:
Also my new 17088572-377F-44EF-8F4E-B09FFF46A070 which is suppose to be 800h bytes larger is inserted back at the original file size incorrectly such that the last microcode is cut short..

Sorry for all the grief.. Its not my bios file but someone elses though I took an interest in seeing whats going on.. Thanks.

davidm71, 2018-01-30:
Also how is the TE_Header size = 28h ?  UEFITool says 4h??

NikolajSchlej, 2018-01-30:
It's a known bug with keeping the original size, will be fixed soon. As for pull request, inclusion of teFixup into relocation (which is spc-compliant but will break XIP TE images on almost everything because IRL almost nothing is spec compliant) makes this PR unmergeable. Other fixes will be integrated, thanks a ton.

davidm71, 2018-01-31:
Coderush man thanks for everything.. understood only ten percent of what you said but as far as what platforms 0.22 is compatible with can we please have some clarification in regards to replacing the Microcode master file and indication for correcting PEI-Core addressing? Thank you.

davidm71, 2018-02-04:
What does this mean:
findFitRecursive: FIT table candidate found, but not referenced from the last VTF ???

davidm71, 2018-02-10:
@serg-pushkarev 
Has these issues been fixed yet in the latest release of UEFITool?

Thanks

serg-pushkarev, 2018-02-10:
I don't know. I am not collaborator, just user.

davidm71, 2018-02-10:
My mistake. Wait for CodeRush then to comment. Thanks.

NikolajSchlej, 2018-02-15:
Should be fixed now, please verify.

davidm71, 2018-02-15:
I can certainly verify that it now works amazingly well and can replace the body with larger + smaller file sizes with no corruption in the rest of the volume. Tested it out on a few bios files with complete success.  Thank you for fixing it so fast. Always a fan!

Love Robot Chicken! Its alive!!!

closed 2018-02-15 by NikolajSchlej