LongSoft/UEFITool #109: PeiCore Entry Point Address request

davidm71, 2018-02-13:
Hi,

When studying the Peicore pointer address at FFFFE0 it is confusing to tell looking in UEFITool what that address is when you select PeiCore-> TE Image/Pe32 Image. I found it simplest to add 'Image Base' address with 'Entry Point' Address to get that number.  Wish UefiTool would figure that out for you and display it on the right side panel. 

Thank you.

NikolajSchlej, 2018-02-13:
ImageBase is an arbitrary value read from PE/TE header (not all images are properly based), and just adding it to EntryPoint makes little sense. Why do need to do anything with PEI core?

davidm71, 2018-02-13:
In regards to adding microcode files  to one's bios where the GUID size may change as I understand it is possible that the PeiCore location may rarely shift position such that according to Serg-pushkarev:
> 

>  You need to perform some actions for the final firmware to work. You have to expect, what final address of PEI_CORE is correct. You need to open the firmware in the hex editor and make sure that the address is set correctly or correct it. (See pics 1 and 2: https://yadi.sk/a/S2Ix1Soy3RtnCn). Good luck! :)
> --
> 

He implied that you have to add the 'StrippedSize' + 'Entrypoint' - headersize to get that address. I found it simplest to add 'Image Base' + 'Entry Point' Address to get that value.  But its confusing for most to understand and would be easier if UefiTool worked out the math for you.

Thank you.

NikolajSchlej, 2018-02-13:
Moving around PEI Core (any PEI images, but PEI core in particular) is very sensitive operation. Sometimes it works, sometimes it breaks everything, even if you fix the pointer to PEI core in Volume top file and rebase everything properly (detecting what "properly" is here is the most difficult task of the whole operation). I've implemented some heuristics back then, but now I see that they aren't really working for lots of images.
Even if PEI core is guaranteed to be based properly (otherwise it won't execute), both of it's ImageBase and EntryPoint values are not used (try changing them and see if it breaks booting), so they may be arbitrary. Showing them will only introduce confusion ("why is that field there and what does it mean?"), and relying on them is wrong (as they may both contain zeros, FFs, or anything else).

NikolajSchlej, 2018-02-13:
My point is, after working on this tool for 5+ years: it's way better to not support something and not allow something, than support it a half-ass way, allowing seemingly harmless operations to break your firmware. PEI core should be fixed in the image, and if microcodes aren't fitting, just move that file with them to the end of PEI volume, replacing the original one with pad-file. Nothing needs to be rebased that way.

davidm71, 2018-02-13:
Thats all very sound advice and I will most certainly listen to your advice and take it to heart.  Thank you.

davidm71, 2018-02-13:
Thanks I am aware of that. Should tell you I am writing a guide detailing the process of manually updating the microcodes here on Win-Raid: https://www.win-raid.com/t3132f47-GUIDE-AMI-INTEL-CPU-Microcode-Update-Guide.html#msg42576.  It would mean a lot to me if you could look it over tell me what you think... Trying not to be too wordy and verbose but there is a lot of information.

Thank you.

closed 2018-03-14 by davidm71