LongSoft/UEFITool #110: Two NVRAM sections

c0d3z3r0, 2018-02-13:
While doing some modifications to my UEFI image I noticed that I have two NVRAM sections.
I thought that must be wrong so I reflashed a virgin UEFI image with an empty NVRAM section only containing StdDefaults via flashrom/SPI. After rebooting the machine some times I read the image back and checked it in UEFITool. Again, there was an additional NVRAM section with identical GUID but two different "State"s (0xF0 vs. 0xE0, virgin image was 0xF8).

Any ideas what is going on here?

Virgin UEFI image:
[image]

Two NVRAM Sections:
[image]
[image]

c0d3z3r0, 2018-02-13:
Ah, looks like I should have waited until the system restarted three times after first power on.
Powering off too early seems to interrupt a setup process... Now there is only one section.

closed 2018-02-13 by c0d3z3r0

NikolajSchlej, 2018-02-13:
The second section is used as backups storage during updates, or as a working storage if the main one fails too hard to be restore by garbage collection. It's ok to have 2 of them.