LongSoft/UEFITool #112: UefiTool 0.22.1 Replace-Body Major Bug discovered !!!

davidm71, 2018-02-14:
CodeRush, Last night I was trying to Replace the body of GUID 17088572-377F-44EF-8F4E-B09FFF46A070 that holds the microcodes for someone on WinRaid. In anycase I had great deal of difficulty using the tool as when you tried to replace-body of the file with one with either a smaller or larger size file, UefiTool 0.22.1 would split the file into the original 17088572-377F-44EF-8F4E-B09FFF46A070, and another GUID such as 
FBB7EC0B-3D19-4623-7C8C-93D5462491BA that contained the tail half of larger file, and then wipe out any modules that followed the intended replace body file. If the replace raw file was smaller 17088572-377F-44EF-8F4E-B09FFF46A070 would retain its original size though have junk code at the end. I was able to solve this by editing the header section pertaining to the FFS file size so I assume UefiTool 0.22.1 is not correcting the header appropriately. In any case 0.21.5 works just fine.

Thank you

Heres a link to the original file for your experimentation: 

https://www.smartredirect.de/redir/clickGate.php?u=CHoN7d6s&m=1&p=3r6MDbeCf4&t=6QdegMhg&st=&s=&splash=2&url=https%3A%2F%2Fmega.nz%2F%23!3xQkzQ7Z!HXnSw7jgtmabnx-9Cl_U4P0OWihHywONOqzJCc42pRs&r=https%3A%2F%2Fwww.win-raid.com%2Ft3132f47-GUIDE-AMI-INTEL-CPU-Microcode-Update-Guide-5.html%23msg47955

NikolajSchlej, 2018-02-14:
Known issue, reported by Sonix a while ago. Will be fixed by project maintainers soon.

davidm71, 2018-02-14:
Well then take 0.22.1 down! Someone is going to get a brick!

NikolajSchlej, 2018-02-14:
I don't have modification rights over this repository. I've asked people who do already, but they are out for now.

Cr4sh, 2018-02-15:
Check latest release please, should work fine
https://github.com/LongSoft/UEFITool/releases/tag/0.22.3

davidm71, 2018-02-15:
Yes it works great now! Thank you very very much!

closed 2018-02-15 by NikolajSchlej