LongSoft/UEFITool #118: UEFITool NEA47 Extract Body name not correctly being identified

davidm71, 2018-03-07:
Hi,

When you extract body of lets say SubGUID 50339D20-C90A-4BB2-9AFF-D8A11B23BC15 with UEFITool NEA47 it will give it the name of the parent GUID A0327FE0-1FDA-4E5B-905D-B510C45A61D0 and that for every other different GUID in the compressed folder you are extracting. This gets confusing as they all end up having the same name. 

Also way too much yellow everywhere. If you want me to mod the colors I would be happy. 

Thank you

NikolajSchlej, 2018-03-07:
Naming is one issue and I'll tell maintainers to check out.

Too much yellow is another and can be trivially solved by disabling BootGuard markings in View menu.

davidm71, 2018-03-07:
Ok. Thank you.

vit9696, 2018-05-06:
Should be fixed in the latest NE version.

closed 2018-05-06 by vit9696