LongSoft/UEFITool #120: UEFITool not working for newer Lenovo images

c0d3z3r0, 2018-03-16:
UEFITool has problems with (some) newer Lenovo UEFI images.
Examples:
ThinkStation P920: https://download.lenovo.com/pccbbs/thinkcentre_bios/s05j319usa.zip
ThinkStation P410/P510: https://download.lenovo.com/pccbbs/thinkcentre_bios/s00j356usa.zip
some others like P720, P520, P520c

P320 works...: https://download.lenovo.com/pccbbs/thinkcentre_bios/s06jt29usa.zip
P900 works, too.

Extracting a LZMA-Section with dd and unpacking it with unlzma or even edk2 decompress does not work, too... "Corrupt data."

Screenshot (P920):
[image]

skochinsky, 2018-03-20:
The image seems to be corrupted around that file. There is a UI and version section  a little later for "MrcOemHooksPeim" but the preceding data with the file body seems to be missing. Maybe the image works by accident (i.e. the modules in the corrupted area are not critical so the boot succeeds). I think something went wrong when the image was built.

c0d3z3r0, 2018-03-20:
It's not only this image but some others, too. I can't believe they're all corrupted... but maybe something went terribly wrong at lenovo..

skochinsky, 2018-03-20:
Actually there is some interesting text at the start:

```
_AMIPFAT.AMI_BIOS_GUARD_FLASH_CONFIGURATIONS
1 /N 1 ;BIOS_FV_NVRAM.bin
1 /WHEA 1 ;BIOS_FV_WHEA.bin
1 /P 11 ;BIOS_FV_MAIN.bin
1 /O 1 ;OEMFV.bin
1 /FIT 1 ;BIOS_FV_FIT.bin
1 /FACM 1 ;BIOS_FV_ACM.bin
1 /B 4 ;BIOS_FV_BB.bin

```

and it is followed by some table of possibly flash block offsets (?). So maybe it needs to be reassembled from pieces or something.

PFAT is "Platform Firmware armoring technology" and is better known under the brand name "BIOS Guard" (NB! different from Boot Guard). Maybe @matrosov knows something...

c0d3z3r0, 2018-03-20:
MMTool says "Image is corrupt", too

c0d3z3r0, 2018-03-20:
@skochinsky Looks like you are right.
I had a short look to AfuEfix64.efi with IDA and found this:

~~~
" Convert BIOS Guard files To Normal File... (%d%%) \r"
...
...
" - Convert BIOS Guard files To Normal File... done \n"
~~~

platomav, 2018-03-22:
UEFITool does not detect PFAT/AMI Bios Guard. So far we have seen two AMI PFAT revisions: 1 and 2.

The header is Size (0x4) + Checksum (0x4).

--- Start of 1st Section ---

After the header we have PFAT Revision (0x4) followed by Platform (0x10).

Then the PFAT Block which is 0x1C in size and consists of Unknown1 (0x4) + Unknown2 (0x4) + Block Size (0x4, from Block Start) + Image Size (0x4, from Block End) + Unknown3 (0x4) + Unknown4 (0x4) + Unknown5 (0x4).

When you go to Image End (Block End + Image Size), you'll find an RSA Block which is 0x20C in size and consists of Signature (0x4) + Flags (0x4) + RSA Public Key (0x100) + RSA Exponent (0x4) + RSA Signature (0x4).

--- Start of 2nd Section ---

We have PFAT Revision (0x4) followed by Platform (0x10).

etc

vit9696, 2018-11-12:
I believe with [AMI BIOS Guard Extractor v1.0](https://github.com/platomav/BIOSUtilities/blob/master/AMI%20BIOS%20Guard%20Extractor/AMI_PFAT_Extract.py) ([binaries](https://github.com/platomav/BIOSUtilities/releases)) by @platomav available, there is little reason for us to reimplement such functionality in UEFITool. After all, it is quite vendor specific.

closed 2018-11-12 by vit9696