LongSoft/UEFITool #122: UEFIPatch not working on laptop ASUS GL552VW(Skylake)

fen3der, 2018-03-17:
Trying to patch the BIOS fails

```
parseImageFile: Aptio capsule signature may become invalid after image modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 7 bytes at offset A9E3h 75080D00800000 -> EB080D00800000
reconstructSection: executable section rebase failed
Error ,
```

BIOS
http://dlcdnet.asus.com/pub/ASUS/nb/GL552VW/GL552VWAS300.zip

vit9696, 2018-03-19:
Works just fine with the latest UEFIPatch.
```
$ UEFIPatch ~/Downloads/GL552VW-AS.300 
parseImageFile: Aptio capsule signature may become invalid after image modifications
parseFile: non-empty pad-file contents will be destroyed after volume modifications
patch: replaced 7 bytes at offset A9E3h 75080D00800000 -> EB080D00800000
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
rebase: can't determine if TE image base is adjusted or not, rebased TE image may stop working
Image patched
$ UEFIPatch 
UEFIPatch 0.3.13 - UEFI image file patching utility

Usage: UEFIPatch image_file

Patches will be read from patches.txt file
```
If you need macOS binaries, they are posted in the dedicated thread on [AppleLife](https://applelife.ru/threads/universalnyj-patcher-uefi-biosov.37815/)

fen3der, 2018-03-20:
compiled versions tools with applelife patch, but after patching the patched bios kernelPm is still required
This means that the patch does not work on this BIOS

vit9696, 2018-03-20:
You came up with an incorrect conclusion. I disassembled your BIOS and checked the stuff. SiInit is pretty much the only place that writes to 0xE2, and the patch is valid. The fact it did not work does only mean you failed to properly flash a modified firmware. This is likely since you do not seem to be using a SPI Flasher.

fen3der, 2018-03-20:
Without SPI Flasher you can get by or is this the only way to flash correctly?

vit9696, 2018-03-20:
Depends, I did not have much experience with different hardware. I suppose these questions are not intended for an issue tracker anymore anyway.

fen3der, 2018-03-20:
Thank you. Close

closed 2018-03-20 by fen3der