LongSoft/UEFITool #132: Unable to compile sources on Windows

juancorradine, 2018-05-06:
Hi.. I'm trying to compile the latest sources in Windows, as the precompiled section only has files for Mac, not Windows.

I downloaded/installed Qt 5.10.1 and Visual Studio Developer Command Prompt. I then ran the commands:
C:\Qt\5.10.1\msvc2017_64\bin\qmake.exe c:\dev\UEFITool-master\uefitool.pro
Got:
Info: creating stash file C:\Dev\UEFITool-master\.qmake.stash

Then on "x64 Native Tools Command Prompt for VS 2017" I typed:
make

I get:
make -f Makefile.Release
make[1]: Entering directory `c:/Dev/UEFITool-master'
Makefile.Release:145: *** missing separator.  Stop.
make[1]: Leaving directory `c:/Dev/UEFITool-master'
make: *** [release] Error 2

Some stackoverflow forums suggested that a tab was missing in that line before << ..
So I added a tab on all the lines that started with <<

Tried to run make again, and it showed the same type of error, but for line 249, which has:
release\uefitool_main.obj release\uefitool.obj release\searchdialog.obj release\types.obj release\descriptor.obj release\ffs.obj release\peimage.obj release\ffsengine.obj release\treeitem.obj release\treemodel.obj release\messagelistitem.obj release\guidlineedit.obj release\LzmaCompress.obj release\LzmaDecompress.obj release\LzFind.obj release\LzmaDec.obj release\LzmaEnc.obj release\EfiTianoDecompress.obj release\EfiTianoCompress.obj release\EfiTianoCompressLegacy.obj release\moc_uefitool.obj release\moc_searchdialog.obj release\moc_ffsengine.obj release\moc_treemodel.obj

I also added a tab on this line, and now I'm getting:

make -f Makefile.Release
make[1]: Entering directory `c:/Dev/UEFITool-master'
Makefile.Release:250: *** multiple target patterns.  Stop.
make[1]: Leaving directory `c:/Dev/UEFITool-master'
make: *** [release] Error 2

I tried also with MinGW, and got the same results.

Here's the complete Makefile.Release: https://pastebin.com/KRgFLRNa

Help :)

vit9696, 2018-05-06:
Use a separate build directory (e.g. build) instead of running stuff in the same directory the sources are.

closed 2018-05-06 by vit9696

juancorradine, 2018-05-06:
Hi vit9696,

Thanks for your response.

I created a separate build directory, and ran the build commands from there:
c:\Dev\UEFITool-master\build>C:\Qt\5.10.1\msvc2017_64\bin\qmake.exe c:\dev\UEFITool-master\uefitool.pro

I get:

Info: creating stash file C:\Dev\UEFITool-master\build\.qmake.stash
WARNING: C:/../dev/UEFITool-master/descriptor.cpp conflicts with C:/dev/UEFITool-master/descriptor.cpp
WARNING: C:/../dev/UEFITool-master/ffs.cpp conflicts with C:/dev/UEFITool-master/ffs.cpp
WARNING: C:/../dev/UEFITool-master/ffsengine.cpp conflicts with C:/dev/UEFITool-master/ffsengine.cpp
WARNING: C:/../dev/UEFITool-master/guidlineedit.cpp conflicts with C:/dev/UEFITool-master/guidlineedit.cpp
WARNING: C:/../dev/UEFITool-master/messagelistitem.cpp conflicts with C:/dev/UEFITool-master/messagelistitem.cpp
WARNING: C:/../dev/UEFITool-master/peimage.cpp conflicts with C:/dev/UEFITool-master/peimage.cpp
WARNING: C:/../dev/UEFITool-master/searchdialog.cpp conflicts with C:/dev/UEFITool-master/searchdialog.cpp
WARNING: C:/../dev/UEFITool-master/treeitem.cpp conflicts with C:/dev/UEFITool-master/treeitem.cpp
WARNING: C:/../dev/UEFITool-master/treemodel.cpp conflicts with C:/dev/UEFITool-master/treemodel.cpp
WARNING: C:/../dev/UEFITool-master/types.cpp conflicts with C:/dev/UEFITool-master/types.cpp
WARNING: C:/../dev/UEFITool-master/uefitool.cpp conflicts with C:/dev/UEFITool-master/uefitool.cpp
WARNING: C:/../dev/UEFITool-master/uefitool_main.cpp conflicts with C:/dev/UEFITool-master/uefitool_main.cpp
WARNING: Automatically turning off nmake's inference rules. (CONFIG += no_batch)
WARNING: C:/../dev/UEFITool-master/descriptor.cpp conflicts with C:/dev/UEFITool-master/descriptor.cpp
WARNING: C:/../dev/UEFITool-master/ffs.cpp conflicts with C:/dev/UEFITool-master/ffs.cpp
WARNING: C:/../dev/UEFITool-master/ffsengine.cpp conflicts with C:/dev/UEFITool-master/ffsengine.cpp
WARNING: C:/../dev/UEFITool-master/guidlineedit.cpp conflicts with C:/dev/UEFITool-master/guidlineedit.cpp
WARNING: C:/../dev/UEFITool-master/messagelistitem.cpp conflicts with C:/dev/UEFITool-master/messagelistitem.cpp
WARNING: C:/../dev/UEFITool-master/peimage.cpp conflicts with C:/dev/UEFITool-master/peimage.cpp
WARNING: C:/../dev/UEFITool-master/searchdialog.cpp conflicts with C:/dev/UEFITool-master/searchdialog.cpp
WARNING: C:/../dev/UEFITool-master/treeitem.cpp conflicts with C:/dev/UEFITool-master/treeitem.cpp
WARNING: C:/../dev/UEFITool-master/treemodel.cpp conflicts with C:/dev/UEFITool-master/treemodel.cpp
WARNING: C:/../dev/UEFITool-master/types.cpp conflicts with C:/dev/UEFITool-master/types.cpp
WARNING: C:/../dev/UEFITool-master/uefitool.cpp conflicts with C:/dev/UEFITool-master/uefitool.cpp
WARNING: C:/../dev/UEFITool-master/uefitool_main.cpp conflicts with C:/dev/UEFITool-master/uefitool_main.cpp
WARNING: Automatically turning off nmake's inference rules. (CONFIG += no_batch)

I then run: 
c:\Dev\UEFITool-master\build>nmake release

And I get (apparently no errors): https://pastebin.com/XAT5AUFL

I then run:
c:\Dev\UEFITool-master\build>make release

And I get:
make -f Makefile.Release
make[1]: Entering directory `c:/Dev/UEFITool-master/build'
Makefile.Release:161: *** missing separator.  Stop.
make[1]: Leaving directory `c:/Dev/UEFITool-master/build'
make: *** [release] Error 2

Line 161 in Makefile.release has:
release\uefitool_main.obj release\uefitool.obj release\searchdialog.obj release\types.obj release\descriptor.obj release\ffs.obj release\peimage.obj release\ffsengine.obj release\treeitem.obj release\treemodel.obj release\messagelistitem.obj release\guidlineedit.obj release\LzmaCompress.obj release\LzmaDecompress.obj release\LzFind.obj release\LzmaDec.obj release\LzmaEnc.obj release\EfiTianoDecompress.obj release\EfiTianoCompress.obj release\EfiTianoCompressLegacy.obj release\moc_uefitool.obj release\moc_searchdialog.obj release\moc_ffsengine.obj release\moc_treemodel.obj

So if I add a tab at the beginning of line 161 in Makefile.release, and run make release I get:

make -f Makefile.Release
make[1]: Entering directory `c:/Dev/UEFITool-master/build'
Makefile.Release:162: *** multiple target patterns.  Stop.
make[1]: Leaving directory `c:/Dev/UEFITool-master/build'
make: *** [release] Error 2

So I'm getting the same error even if I do it in a separate build directory.

Any clues as to what might I be doing wrong?

Thanks!

vit9696, 2018-05-06:
Feels very strange, but I somehow get a feeling it might be a Qt bug. I do not have Windows nearby at the moment, and if nmake works for you it will at least behave as a workaround. I will try to get my hands onto Windows sooner or later and see if I can reproduce it myself.

vit9696, 2018-05-06:
Ugh, I saw your Makefile, and it looks like $(LIBS) is on the next line for whatever reason, put it to the previous line and hopefull iy will work.

juancorradine, 2018-05-07:
Ok, getting a bit further. I did the change you suggested, and also had to add a few tabs.

Here's how the makefile looks like now: https://pastebin.com/49yM2WxE

And here's the output: https://pastebin.com/30KqrzXa

Error:

link /NOLOGO /DYNAMICBASE /NXCOMPAT /INCREMENTAL:NO /SUBSYSTEM:WINDOWS "/MANIFESTDEPENDENCY:type='win32' name='Microsoft.Windows.Common-Controls' version='6.0.0.0' publicKeyToken='6595b64144ccf1df' language='*' processorArchitecture='*'" /MANIFEST:embed /OUT:release\UEFITool.exe @<<
<< was unexpected at this time.
make[1]: *** [release\UEFITool.exe] Error 255
make[1]: Leaving directory `c:/Dev/UEFITool-master/build'
make: *** [release] Error 2

By the way, I was able to use the exe generated by nmake.. Just curious why make is not working..

vit9696, 2018-05-07:
What I see there does not even feel like a valid make language to my eyes. From that I can conclude that this code was completely untested with this configuration (and/or meant to be executed with anything but gnu make). You could probably get further by changing it like this (with the removal of last <<):

```
$(LINKER) $(LFLAGS) /MANIFEST:embed /OUT:$(DESTDIR_TARGET) $(LIBS) release\uefitool_main.obj release\uefitool.obj release\searchdialog.obj release\types.obj release\descriptor.obj release\ffs.obj release\peimage.obj release\ffsengine.obj release\treeitem.obj release\treemodel.obj release\messagelistitem.obj release\guidlineedit.obj release\LzmaCompress.obj release\LzmaDecompress.obj release\LzFind.obj release\LzmaDec.obj release\LzmaEnc.obj release\EfiTianoDecompress.obj release\EfiTianoCompress.obj release\EfiTianoCompressLegacy.obj release\moc_uefitool.obj release\moc_searchdialog.obj release\moc_ffsengine.obj release\moc_treemodel.obj
```
But it looks insane anyway. This is clearly not our fault, and qmake just generates garbage.