LongSoft/UEFITool #136: UEFIFind и UEFIExtract использование в скриптах

LSSoniX, 2018-06-02:
Всем доброго здравия.

При использовании данных утилит в скриптах возникают некоторые неудобства
1 - UEFIFind
В коммандной строке мы задаем 
`UEFIFind <аргументы> файл биоса`
Если задавать вручную на раз два или в скрипте всего 2-3 строки, то не всё так плохо. Но если в сурипте надо задать десятки раз, то как то не совсем удобно описывать всю строку.
Думаю, что вариант
`UEFIFind файл биоса <аргументы>`
был бы самыс оптиальным.
В скриптах уже можно будет задать переменную, например
`varuf UEFIFind bios.bin`
и далее задавать только переменную
`varuf <аргументы>`
что согласитесь гораздо удобнее и потом общий формат команной строки для всех консольных утилит линейти UEFITool.

2 - UEFIExtract.
Очень сложно испольщовать ее в скриптах, порой даже не представляю как.
Утилита очень хороша, находит, что нужно и разбирает всё по полочкам, но вот в скриптах сложно задавать пути к разобранным файлам.
Возможно ли добавить еще аргументы для извлечения файла целиком из конкретно GUID илм SubbGШD в указанное место? Напимер
`UEFIExtract файл биоса <GUID|SubGUID> -o имя выходного файла`
при этом есть одна критичность, если FFS содержит в себе сжатие, то при извлечении желательно чтобы выходной файл был рассжат, что даст возможность обработать его в дальнейшем его обработать.
Задание SubGUID для извлечение из FreeForm.

Других замечаний нет и всем огромное спасибо. :))

vit9696, 2018-06-02:
Привет,

master ветку зарелизил как 0.25.0, по существу там изменений нет по сравнению с тем, что давал тебе (кроме номера).

1. Исправил, действительно не очень логично, что во всех утилитах один порядок, а здесь другой.

2. Попробовал исправить, насколько можно. Прототип следующий. В программе есть набор управляющих аргументов:
- `-i` — input, входные файлы
- `-o` — output, выходная папка
- `-m` — mode, режим извлечения
- `-t` — type, тип секции
После каждого флажка идёт список значений, который должен быть равен количеству входных файлов. `-i` опционален и подразумевается по умолчанию (для сохранения совместимости интерфейса). Передача каких-либо флажков (и аргументов к ним кроме входных файлов) также опциональна.

Режим извлечения: сейчас и раньше по умолчанию all. Можно также указать body (для извлечения section body), header, а также info (для информационного файла).

Тип секции: шестнадцатеричное значение, как и в UEFIReplace. Использовать тогда, когда в ffs несколько секций разного типа. Если надо извлекать более одного файла, и для части ffs тип секции неизвестен или режим, к примеру, all, то можно указать FF для игнора (небольшой костыль).

Выходная папка указывается как и раньше, однако в режимах body, header, info файл будет создан непосредственно в данной папке без иерархии (с именем body.bin, header.bin, info.txt).

Попробовать можно здесь: https://github.com/LongSoft/UEFITool/releases/tag/t20180602

LSSoniX, 2018-06-03:
Протестил UEFIFind - не работает, те ничего не находит или падает.
uefifind bios.bin body list паттерн - падение.
uefifind bios.bin all list паттерн - не падает, но и не находит
если искть GUID
uefifind bios.bin heder count паттерн GUID'а - не падает, но опять ничего не находит.
хотя искомые патты есть.

LSSoniX, 2018-06-03:
Протестил UEFIFind - не работает, те ничего не находит или падает.
uefifind bios.bin body list паттерн - падение.
uefifind bios.bin all list паттерн - не падает, но и не находит
если искть GUID
uefifind bios.bin heder count паттерн GUID'а - не падает, но опять ничего не находит.
хотя искомые патты есть.
Падение выражается сообщееикм, что утиля не работает, точно такое же когда реплейсер тестили.

vit9696, 2018-06-03:
Там Qt старый… Отправил коммит с фиксом, скоро пересоберётся.

LSSoniX, 2018-06-03:
Проверил пофиксенный УФ, работает. Спасибо. :)

Хорошо, так и сделаю.
PS Экстратором попозже займусь, глаза уже болят..

vit9696, 2018-06-03:
Хорошо, кстати, а ты не смотрел в сторону UEFIDump? У него формат распаковки куда более удобен и никаких аргументов особо не надо.

LSSoniX, 2018-06-08:
Всем доброго здравия!
Прошу меня извинить за долгий ответ.

UEFIExtract
Опробовал новые опции, к сожалению, как я и предполагал, проще не стало.
По сути мы получили расширение уже имеющего функционала, ео не более. Если чесно я вообще не вижу в нем смысла, хотя может кому то понравится.

Как я уже указывал оптимальный вариант это
"Extract file as is with DeCompressed if exist Compressed to MyPath\MyName"
Вот тут пример
https://cloud.mail.ru/public/AQLL/XcaaDvsuv
Мы находим нужный файл, получаем его GUID и извлекаем как есть. При этом если файл имеет сжатие, то производим его рассжатие. Результат сохраняем в указанную нами папку и своим имени.

vit9696, 2018-06-08:
Привет, из примера ничего не понятно, у нас всё работает, а новые опции обеспечивают полное выполнение функционала.

Конкретно сохранять *файл* под именем никто не будет, это на логику программы не ложится, только папку. И это правильно, так как контролируется рабочее пространство программы, если папка уже создана, то будет ошибка.

Узкоспециализированного выдрать "body по guid, при этом разжав его и переименовав" тоже не будет, так как это не решает большинство других use-case.

У меня работает:
```
./UEFIExtract bios.bin -i 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o SataDriver -m body -t 10
```

LSSoniX, 2018-06-08:
А если тип секции нам не известен? Или в FFS несколько разных типов секций?
Я привел пример с MyName1 извлеченного полного FFS как есть со всем содержимым в одном флаконе, а так как он в имидже сжат, то при сохранении выплнено рассжатие. И это правильно, тк можно сразу видеть всё что есть в одном файле. Те варианты, что есть сейчас они хороши для визуального изучения, а не скриптового.

vit9696, 2018-06-08:
Так, ты меня сейчас заставишь делать очередную ненужную фичу, потому что мы говорим на разных языках. Код я пишу постоянно, и откровенно неюзабельный для скриптов инструмент бы не делал. Если мы готовим его по-разному, то это не значит, что он плох.

Смотри, есть разница между тем, какая задача и что ты хочешь сделать. Причём последнее меня интересует меньше всего, потому что это следствие, а оно продиктовано другими инструментами. Мне бы хотелось получать информацию так:
1. Есть прошивки (например, XXX.bin, YYY.bin, ZZZ.bin)
2. Надо в них заменить драйвер САТА на более новый, в XXX он находится под GUID 12345, в YYY под 45678…
3. Я ищу драйвер SATA с помощью паттерна 12345, потому что так проще, а GUID отличаются в каждой прошивке. Для этого выполняю следующую команду.
…

Т.е. описывай мне задачу максимально абстрактно, но при этом указывай, как ты её решаешь, и почему ты так делаешь. Если ты будешь писать только то, как ты пишешь своё решение, ничего хорошего не выйдет, так как глупые фичи я пилить просто не буду.

По твоему второму use-case, про поиск SubGUID я просто забыл, теперь он работает.
```
./UEFIExtract bios.bin  -i 380B6B4F-1454-41F2-A6D3-61D1333E8CB4 -o 2ndcase -m body
```

По поводу неизвестных типов секций — не ясен use-case, делать гипотетические фичи никто не будет. Для разных типов секций разные номера и можно указывать. Я в принципе могу добавить извлечение FFS, но тебе придётся меня убедить, что это реально жутко необходимая фича. В моём понимании FFS надо всегда пересобирать, потому извлекай нужные body нужных секций,  затем формируй новый ffs, и его уже используй.

LSSoniX, 2018-06-09:
Вполне вероятно, что я недопонимаю концпцию работы УЕ. И для меня весьма странно, что в GUI УТ я могу установить курсор на нужный GUID щелкеуть ПКМ и выбрав "Extract as is" извлечь всю FFSлу, но не могу сделать тоже самое в консольной версии УЕ.
И боже упаси, никого не завставляю, что менять радикально. Если есть возможность внести - хорошо, если нет - то значит нет будем искать другие методы. :))

Хорошо. На базе того, что мы имеем есть некоторые решения, но у меня возникает 3 нюанса, которые хотелось бы разобрать и закрыть тему. Надеюсь, что найдется время помочь? :))

platomav, 2018-06-11:
Hi,

Great work @vit9696. I noticed that UEFIExtract extracts by default the last GUID, if multiple exist, and not the first. Here's what I'm doing:

`UEFIFind uefi.bin body list pattern`

It lists 5 GUIDs which include the pattern, 4 x GUID_A and 1 x GUID_B. There are multiple GUID_A but I am interested in the 1st. So at UEFIExtract I input:

`UEFIExtract uefi.bin GUID_A -o output -m body -t FF`

It works but outputs the 4th GUID_A, not the 1st. It's like it extracts in reverse, from bottom to top.

vit9696, 2018-06-12:
Hmmm, I could consider implementing some sort of Nth flag or suffix the extractions, but both approaches feel like hacks to my eyes. What is the exact use case? Extracting the first or the last is kind of not right thing to do, what if one needs the middle one?

platomav, 2018-06-12:
I have an Apple EFI and want to detect its version tag. To do that I need to search for which GUID contains the pattern $IBIOSI$ and then extract it to retrieve the info from it via a python script.

[image]

That results in an "ibiosi" folder which contains a single body.bin, as expected. Thing is, the first 3 C3E36D09-8294-4B97-A857-D5288FE33E28 contain the proper data but the 4th (last) has different/wrong data. UEFIExtract extracts the last one and that's where the problem is.

Generally, I think that the logical thing is to extract the 1st result, not the last. That won't solve the problem for everyone but it makes more sense I suppose.

[efi.zip](https://github.com/LongSoft/UEFITool/files/2092131/efi.zip)

vit9696, 2018-06-12:
Ok, I changed the logic to save all the subsequent extraction with a numeric suffix (i.e. body.bin, body_1.bin, body_2.bin). Should do for most of the cases.

platomav, 2018-06-12:
Ah, perfect. I tested it and works as expected. Thank you vit9696. Your implementation is even better than an index because someone may not know which exact GUID holds what they're looking for so, by extracting all of them, external logic can be used to easily determine that after UEFIExtract.

LSSoniX, 2018-06-13:
Привет.
vit9696, извини, долго собирался с мыслями. В общем по тестам у меня получается такакя картина.
Используются последние сборки, кстати ситуация чуток схожа с послденими сообщениями.
И так. У нас есть некий список паттернов которые надо найти и извлечь все найденные файлы в одну папку, например в ТМР, для последующей обработки.
Рисуем простенький скрипт (знаю, что виндовый не катит, но башу мне пока не задействовать)

```
1 @echo off
2 pushd %~dp0
3 set ufbl=uefifind bios.bin body list
4 set ue=uefiextract bios.bin
5 for /f "eol=# tokens=*" %%a in (_List_Pattern.txt) do (
6 for /f "tokens=1,2" %%b in ('%ufbl% %%a') do (
7	set SubGUID=%%c
8	if not defined SubGUID (
9	%ue% %%b -o tmp -m body -t 10 && move tmp\body.bin tmp\file_%%b.tmp
10	) else (
11	%ue% %%c -o tmp -m body -t 18 && move tmp\body.bin tmp\file_%%c.tmp
12 )))
```
В результате получаем результат в скрине
[image]

но если сделать 9 и 11 строку так
```
9 %ue% %%b -o tmp0 -m body -t 10 && move tmp0\body.bin tmp\file_%%b.tmp && rd /s /q tmp0
11 %ue% %%b -o tmp0 -m body -t 10 && move tmp0\body.bin tmp\file_%%b.tmp && rd /s /q tmp0
```
то всё работает как надо. Извлеченые все файлы в одной папке м у каждого файла своё ммя
В принципе проблем особых нет, но почему нельзя всё делать, в одной папке?

LSSoniX, 2018-06-13:
Забыл. Ткстовый набор тут https://cloud.mail.ru/public/AQLL/XcaaDvsuv

vit9696, 2018-06-13:
Не, ну тут всё как я сказал выше. Защита от дурака, чтобы в существующую директорию ничего не писалось. Это лучше, чем если в совершенно левой папке перезапишется куча файлов. rmdir на команду после копирования считаю вполне приемлемым в подобной ситуации.

LSSoniX, 2018-06-13:
Ну в принципе логично, хотя принцип "сам себе Буратино" интереснее. ;))
Хорошо, с этим нюансом понятно, комманду удаления промежуточной папки можно и в переменную оформить м вызывать когда надо. :)

Второй нюанс у меня с GUID 17088572-377F-44EF-8F4E-B09FFF46A070? это контейнеры микрокодов.
Как ты помнишь это FFS RAW.
В принципе извлечь их на проверку легко, достаточно кузать -m body и даже тип указывать не нлдо.  
Но вот проверить извлеченное на "пустышку" и наличия checksum становится сложнее.
И особенно вернуть "пустышку" обратно в том виде какой она была.
Может есть какая то возможность сделать для таких простеньких FFS RAW извлечение как есть?

LSSoniX, 2018-06-13:
По второму нюансу, FFS RAW (микоды), если нет решения извлекать их как есть, то у меня есть решение обойти всё стандартными средствами самой ОСи. 
`copy tmpmc\header.bin+tmpmc\body.bin mc.ffs`
Но тут опять пляски с бубном возникают.
`uefiextract bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 - o tmpmc -m all`
создаются дебри папок, что понятно не вариант цеплять файлы.
`uefiextract bios.bin 17088572-377F-44EF-8F4E-B09FFF46A070 - o tmpmc -m header -m body`
прнятно, что не работает..
Извлекать всё содержимое, а там всего header и body, отдельно в 2 разные папки за 2 прохода это как то не серьезно.

LSSoniX, 2018-06-13:
Извини. :))
Да, то что полностью забито FF мы называем "пустышкой".
Немного занудной истории из УРЖ
1) Ищем все GUID 1708.......
2) Извлекаем найденный
3) Смотрим его содержимое
- если "пустышка" то заменяем файлом с GUID 1708.......01, а пустышку сохраняем в отдельный файл cpu_zero.ffs
- если не "пустышка" то заменяем файлом с GUID 1708.....02
4) Собираем новый FFS с микодами и заменяем все GUID 1708.....02
5) Возвращаем ранее сохранненую "пустышку" (сpu_zero.ffs) вместо GUID 1708......01
Так вот с последним пунктом у меня проблема, так как на руках то у меня только body "пустыщки", а не полный ffs и вбивисе GUID не родной для замены body -t 1.
Если со сборкой FFS микодов проблем нет, то с "пустышкой" есть заморочки. Встречается так, что размеры этих "пустышек" бывают разеные, а некоторые даже требуют checksum. И чтобы не городить огород я стараюсь вернуть то что было. И получается, что надо вытягивать или весь файл чтобы вернуть потом обратно, или 2 файла header и body - собрав их потом в один целый. 

-m ffs? То есть по сути "asis"? Если конечно это не сложно и возможно, и не будет вытаскивать все разом найденные в разных томах, то да. Спасмбо. :))

vit9696, 2018-06-13:
Добавил, только имя не ffs, а file. Так понятнее. Да, идея в asis.

LSSoniX, 2018-06-13:
Спасибо огромное! :))
По третьему нюансу,, это работа с хранилищем OROMок, пока подумаю детально
А так у меня всё, что, где и как уже понятно как рисовать. :))

Пока теститил, обратил внимание на последних сборках, если задать [-m info] файлы info имееют расширение txt, а если задать [-m all], то файлы info получают расширение bin.

PS у меня к тебе пара личных вопросов есть по иакоси да баше, я их в личку на оверах обрисую попозже.

vit9696, 2018-06-13:
Хорошо, но лучше на applelife где-нибудь, потому что на оверклокерсе какая-то дичь с банерорезками и мне туда заходить проблематично.

closed 2018-06-13 by vit9696

LSSoniX, 2018-09-22:
Всем доброго здравия!
UEFIExtract NE A51 с момента релиза накопилось несколько нюансов которые хотелось бы прояснить.
Тестовый файл биоса https://cloud.mail.ru/public/64Lc/8vuiN6yVU
1) опция '-о имя папки'
Например распаковываем файл PE32 IRST в папку Test
`uefiextract ROG-RAMPAGE-VI-EXTREME-ASUS-1401.CAP 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o test -m body -t 10`
В результате в папке test видим файл body.bin и errorlevel по выполнению 0.
Теперь сделаем "ошибку" в -t
`uefiextract ROG-RAMPAGE-VI-EXTREME-ASUS-1401.CAP 91B4D9C1-141C-4824-8D02-3C298E36EB3F -o test -m body -t 18`
В результате получаем пустую папку test и ! errorlevel 0 !, что не есть правильно. По логике если не найдено хоть что то папка не должна быть создана и левел не должен быть 0.

2) опция '-m file'
К примеру извлечем все FFS-RAW микодов
`uefiextract ROG-RAMPAGE-VI-EXTREME-ASUS-1401.CAP 17088572-377F-44EF-8F4E-B09FFF46A070 -o test -m file`
Всё хорошо и замечательно все 3 файла как есть, но в биосе есть и дркгие файлы собранные как FFS RAW и вот с ними проблема, например в этом же биосе OROM VBIOS. Извлекаем:
`uefiextract ROG-RAMPAGE-VI-EXTREME-ASUS-1401.CAP B981F167-9D2B-4CA0-82E9-63DF0E3C3BF3 -o test -m file`
В папке test получаем file.ffs, но размер явно не htader+OROM, те 65 кило а целый DXE  том.

Есть еще проблемы, но они связаны с измененияии в этом сообщении https://github.com/LongSoft/UEFITool/issues/136#issuecomment-396606346
Я не совсем понял суть этих изменений, тк всё начало работать как то не так..
Если можно пояснить суть.
Спасибо. :)